riscv-software-src/riscv-isa-sim #253: std::runtime_error: couldnt allocate ... of target memory

eleven121, 2018-10-26:
Hi
    I have built the risc-v tools in the ubuntu16.04, and trying to simulating the simple helloworld program using riscv64-unknown-elf-gcc, and the spike and pk: spike pk hello. but i get this problem:
terminate called after throwing an instance of 'std::runtime_error'
what(): couldn't allocate 2147483648 bytes of target memory
Aborted (core dumped)
please help me!

aswaterman, 2018-10-26:
The default is to model a 2 GiB memory, and your host system doesn’t have enough. Try passing the -m256 flag to a spike to reduce that to 256 MiB.

eleven121, 2018-10-26:
spike -m128 pk test
the same problem: https://github.com/riscv/riscv-tools/issues/230
thanks @aswaterman

aswaterman, 2018-10-26:
You’re welcome

closed 2018-10-26 by aswaterman