riscv-software-src/riscv-isa-sim #254: Spike dependency on fesvr

heshamelmatary, 2018-10-27:
Currently, Spike can't be built without fesvr (even though there is a _--without-fesvr_ flag). There are also issues, that don't frequently get triggered, when a specific version of Spike doesn't build with another (older or newer) fesvr revisions. 

My question is whether Spike is expected, at all, to be built without fesvr (and why?):

1- If the answer is yes, what's the correct way to decouple Spike from fesvr?

2- If the answer is no, does it make sense to make fesvr a submodule of Spike and build both in the same step?

damaxi, 2018-11-27:
Maybe we can remove that flag. I also found that fesvr library is quite important here.
#include <fesvr/htif.h>
#include <fesvr/context.h>
#include <fesvr/option_parser.h>
Option parser is heavy used at the beginning and htif seems too.

aswaterman, 2018-11-27:
I support the goal of making spike a library that's independent of fesvr.  In that scenario, the current spike executable would depend upon both libspike and libfesvr, but libspike would be independently useful.

This refactoring would be rather invasive, and there's currently not very high on the list.