riscv-software-src/riscv-isa-sim #256: how dose spike exit??

IC-Dream, 2018-11-08:
Hi, 

I run a ELF that is from riscv/riscv_tests  in spike ang got a commit log, and I noticed that the log enter a loop of write_tohost in the end.  I want to ask how does the spike exit, or implement a watchdog for exit ? could you please tell me the flow of spike exit. 

Since i got a issue that the spike hangs when run my ELF in spike and i didn't implement tohost and fromhost symbols in my ELF.

Thanks a lot
Dream

IC-Dream, 2018-11-12:
I added the tohost and fromhost symbols in .ld files and jump to write "1" to tohost in the end of test case. the spike can exit successfully.

closed 2018-11-12 by IC-Dream

byllgrim, 2020-02-25:
Hi,
sorry to necrobump, but I feel the lack of documentation justifies it.

For anyone else having similar troubles:
> I added the tohost and fromhost symbols in .ld files and jump to write "1" to tohost in the end of test case

This ^ is true.
_But_ not if you run interactively.

You might get a trap_instruction_access_fault etc etc and spend maaany hours thinking that something else is wrong (ahem).

```
.section .text
.global _start

_start:
        la      t1, tohost
        li      t2, 1
        sw      t2, 0(t1)

.section .tohost, "aw", @progbits
.globl tohost
.align 4
tohost: .dword 0
.globl fromhost
.align 4
fromhost: .dword 0
```

This works for me.

Manahil14, 2022-06-06:
This is my startup.S file's code. If we change it according to your solution, will it affect my core's functionality?

    .section .reset, "ax"
    .global _start
_start:
    la      t0, trap_handler
    csrw    mtvec, t0
    csrrsi  zero, mtvec, 1      // set vector based trap

    la      t0, _bss_start
    la      t1, _bss_end

_bss_clear:
    sw      zero,0(t0)
    addi    t0, t0, 4
    bltu    t0, t1, _bss_clear

    la      sp, _stack
    call    main
    tail    exit

    .section .text.trap, "ax"
    .align 4
trap_handler:
    jal     x0, _trap_handler       // 0: Instruction address misaligned
    jal     x0, _trap_handler       // 1: Instruction access fault
    jal     x0, _trap_handler       // 2: Illegal instruction
    jal     x0, _trap_ebreak        // 3: Breakpint
    jal     x0, _trap_handler       // 4: Load address misaligned
    jal     x0, _trap_handler       // 5: Load access fault
    jal     x0, _trap_handler       // 6: Store/AMO address misaligned
    jal     x0, _trap_handler       // 7: Store/AMO access fault
1:  jal     x0, 1b                  // 8:
1:  jal     x0, 1b                  // 9:
1:  jal     x0, 1b                  // 10:
    jal     x0, _trap_ecall         // 11: Environment call from M-mode