riscv-software-src/riscv-isa-sim #260: getenv sigsegv

LMMilewski, 2018-11-27:
Executing the following program:

```
#include <stdio.h>
#include <stdlib.h>
int main(int argc, char** argv) {
  printf("%s\n", getenv("LANG"));
  return 0;
}
```
results in

> z  0000000000000000 ra 0000000000010424 sp 000000007f7e9ad0 gp 0000000000014860
> tp 0000000000000000 t0 0000000000000000 t1 0000000000000000 t2 0000000000000000
> s0 0000000000013108 s1 0000000000000000 a0 0000000000000000 a1 0000000000000000
> a2 00000000000120e4 a3 0000000000000001 a4 000000000000003d a5 0000000000000001
> a6 000000000000001f a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
> s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
> s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
> t3 0000000000000000 t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
> pc 000000000001049c va 0000000000000000 insn       ffffffff sr 8000000200046020
> User load segfault @ 0x0000000000000000
> 

I run the program as

> riscv64-unknown-elf-gcc -march=rv64gc q.c -o q   && spike pk q

It looks like envp is not set at _start (0x100b0):
> spike -d pk ./q
> : until pc 0 100b0
> : reg 0
> zero: 0x0000000000000000  ra  : 0x0000000000000000  sp  : 0x000000007f7e9b50  gp  : 0x0000000000000000  
> tp  : 0x0000000000000000  t0  : 0x0000000000000000  t1  : 0x0000000000000000  t2  : 0x0000000000000000  
> s0  : 0x0000000000000000  s1  : 0x0000000000000000  a0  : 0x0000000000000000  a1  : 0x0000000000000000  
> a2  : 0x0000000000000000  a3  : 0x0000000000000000  a4  : 0x0000000000000000  a5  : 0x0000000000000000  
> a6  : 0x0000000000000000  a7  : 0x0000000000000000  s2  : 0x0000000000000000  s3  : 0x0000000000000000  
> s4  : 0x0000000000000000  s5  : 0x0000000000000000  s6  : 0x0000000000000000  s7  : 0x0000000000000000  
> s8  : 0x0000000000000000  s9  : 0x0000000000000000  s10 : 0x0000000000000000  s11 : 0x0000000000000000  
> t3  : 0x0000000000000000  t4  : 0x0000000000000000  t5  : 0x0000000000000000  t6  : 0x0000000000000000  
> : mem 0 0x000000007f7e9b50
> 0x0000000000000001
> : mem 0 0x000000007f7e9b58
> 0x000000007f7e9bfc
> : mem 0 0x000000007f7e9b60
> 0x0000000000000000
> : mem 0 0x000000007f7e9b68
> 0x0000000000000000
> 

So envp doesn't seem to be initialized. This is consistent with the following program printing 0
```
#include <stdio.h>
int main(int argc, char** argv, char **envp) {
  printf("%p\n", envp);
  return 0;
}
```

I have spike at aff796dbf6db66a2df53b0ca270382f0ce02da74

aswaterman, 2018-11-27:
Spike doesn't implement any of the C library stuff; that's all implemented in the C library you've linked against, which appears to be Newlib.

Seems like the RISC-V Newlib port doesn't support `getenv`.  This is not surprising, since the RISC-V Newlib port is meant for systems without a Unix-like OS.  My advice is to not rely upon `getenv` unless you're really running on a Unix-like system.

closed 2018-11-27 by aswaterman

LMMilewski, 2018-12-02:
It's not clear to me how C library is relevant here (other than getenv being a part of libc). Could you explain that part? I'm sorry if I'm missing something obvious.

I argue that the program is *loaded* without the environment information which I see as a bug. The stack at the beginning of _start (i.e. before any libc code is executed since this is a statically linked binary) looks like:
argc
argv[0]
0
0   /// <- this should be a pointer to envp[0]

The kernel loads the program. That includes putting environment on the stack. Some part of spike provides kernel functionality and my argument is that it should put the environment on the stack.

As far as I understand, no libc code is run before _start in this case, and libc should have no effect on the missing environment.

I'm ok with closing this issue. It's not blocking me, I was exploring riscv for fun and run into this. I run spike on Fedora. I would appreciate it if you explained a little bit more.