riscv-software-src/riscv-isa-sim #262: error: ‘READ_RD’ was not declared in this scope

hepaad, 2018-12-02:
I have the following error:

../riscv/insns/fmac_s.h: In function ‘reg_t rv64_fmac_s(processor_t*, insn_t, reg_t)’:
../riscv/insns/fmac_s.h:12:19: error: ‘READ_RD’ was not declared in this scope
 WRITE_FRD(f32_add(READ_RD(insn.rd()),tmp));

I try to create a new float instruction. fmac.h
This is my code.

require_extension('F');
require_fp;
softfloat_roundingMode = RM;
float32_t tmp = f32_mul(f32(FRS1), f32(FRS2));
WRITE_FRD(f32_add(READ_RD(insn.rd()),tmp));
set_fp_exceptions;

Could you help me to check what happens? 

Regards
Adolfo Hernandez.

aswaterman, 2018-12-02:
It looks like you typed READ_RD instead of READ_FREG.

hepaad, 2018-12-02:
Hi Waterman:

Thank for answer quickly. :)

I tried your solution but I have the same problem 

./riscv/insns/fmac_s.h: In function ‘reg_t rv32_fmac_s(processor_t*, insn_t, reg_t)’:
../riscv/insns/fmac_s.h:13:43: error: could not convert ‘p->processor_t::get_state()->state_t::FPR.regfile_t<float128_t, 32, false>::operator[](insn.insn_t::rd())’ from ‘const float128_t’ to ‘float32_t’
 WRITE_FRD(f32_add(READ_FREG(insn.rd()),tmp));
                                           ^
../riscv/decode.h:171:57: note: in definition of macro ‘DO_WRITE_FREG’
 #define DO_WRITE_FREG(reg, value) (STATE.FPR.write(reg, value), dirty_fp_state)
                                                         ^~~~~
../riscv/decode.h:172:26: note: in expansion of macro ‘WRITE_FREG’
 #define WRITE_FRD(value) WRITE_FREG(insn.rd(), value)
                          ^~~~~~~~~~
../riscv/insns/fmac_s.h:13:1: note: in expansion of macro ‘WRITE_FRD’
 WRITE_FRD(f32_add(READ_FREG(insn.rd()),tmp));
 ^~~~~~~~~
../riscv/insns/fmac_s.h: In function ‘reg_t rv64_fmac_s(processor_t*, insn_t, reg_t)’:
../riscv/insns/fmac_s.h:13:43: error: could not convert ‘p->processor_t::get_state()->state_t::FPR.regfile_t<float128_t, 32, false>::operator[](insn.insn_t::rd())’ from ‘const float128_t’ to ‘float32_t’
 WRITE_FRD(f32_add(READ_FREG(insn.rd()),tmp));
is it possible that I have another issue? 

And another question about the Opcode. 
I chose the following. 
fmac.s    rd rs1 rs2      31..27=0x02 rm       26..25=0 6..2=0x17 1..0=3
In special 6..2 position. What do you recommend that I can select? (I checked that only was different from another.)

Regards
Adolfo Hernandez.

aswaterman, 2018-12-02:
You're probably missing an f32() cast around the READ_FREG.  I suggest studying decode.h and the other FP instructions and figuring this out on your own.

I don't have any opcode recommendations - the ISA manual indicates which opcodes are for custom use.  You can use one of those.

I'm also a little confused why you wouldn't just use the existing fmadd instruction for this purpose?

hepaad, 2018-12-04:
Hi Andrew:

Thank you so much for your help I can create my own instruction with your help.

The problem was how to call into my c file. 

fmac_s.h
require_extension('F');
require_fp;
softfloat_roundingMode = RM;
float32_t tmp = f32_mul(f32(FRS1), f32(FRS2));
WRITE_FRD(f32_add(f32(FRD),tmp));
set_fp_exceptions;

In c file

float fmac_s_asm(float a, float b, float c) {
    asm __volatile__ (".word 0x30C58553\n");
    return a;
}

Regards
Adolfo Hernandez.