riscv-software-src/riscv-isa-sim #268: spike cannot find pk

mahsoommoosa42, 2019-01-01:
Spike cannot find pk. 
`spike pk hello`

returns 

`terminate called after throwing an instance of 'std::runtime_error'
  what():  could not open pk (did you misspell it? If VCS, did you forget +permissive/+permissive-off?)
`

However, 
`spike $(which pk) hello`
works

xiaoxiaohuixxh, 2019-01-15:
you need to make and install pk from https://github.com/riscv/riscv-pk

ilia-diachkov, 2019-03-07:
Install pk in the same place as gcc (your riscv-gnu-toolchain build).

mukesh-cst, 2019-03-10:
> Install pk in the same place as gcc (your riscv-gnu-toolchain build).

I tried installing pk at both the places "/opt/riscv/bin" and "/opt/riscv/riscv64-unknown-linux-gnu/bin". Both the paths has been added to PATH variable. Still, I'm facing this issue.
Although, the above command 'spike $(which pk) hello' works for me.

aswaterman, 2019-03-10:
Try /opt/riscv/riscv64-unknown-elf/bin (assuming you configured riscv-fesvr
with --with-prefix=/opt/riscv)

mukesh-cst, 2019-03-14:
@aswaterman 
Adding the --target=riscv64-unknown-linux-gnu while configuring the riscv-fesvr worked for me.
I installed riscv toolchain while enabling multilib. So, it has installed the libraries under 'opt/riscv/riscv64-unknown-linux-gnu' directory and there is no '/opt/riscv/riscv64-unknown-elf' created for me. 
Thanks for the hint...

aswaterman, 2019-03-14:
Ah - makes sense. Glad you figured it out.

pcotret, 2020-07-15:
I'm still unsure about the best solution to solve this issue...
- Installed the `riscv-gnu-toolchain` in `/opt/riscv`
- Added `/opt/riscv/bin` folder to my `PATH`
- Compiled Spike with `prefix=/opt/riscv/`

`pk` is installed in `/opt/riscv/riscv64-unknown-elf`.
Even if `/opt/riscv/riscv64-unknown-elf/bin` is in my `PATH`, Spike still can't find `pk` unless I specify the path (`spike $(which pk) hello`)

ptprasanna, 2022-02-04:
Found the same issue as @pcotret while running spike.

sansarselim, 2022-02-15:
@pcotret , @ptprasanna 
You need to rebuild all files after changing configure command. Make clean first. After i did that it worked for me. If you configure spike with --prefix=/opt/riscv , spike search files in /opt/riscv/riscv64-unknown-elf/bin/ .

frdpnl, 2025-10-08:
What worked in my case was:
```
$ ../configure --prefix=$RISCV --with-target=riscv64-unknown-linux-gnu
$ # ... (make -j, install)
$ ls $RISCV/riscv64-unknown-linux-gnu/bin/
bbl  dummy_payload  pk
```
(I was missing the `--with-target` option)

Spike kept using the default value: `riscv64-unknown-elf`.
config.h, config.log, config.status all had TARGET_ARCH set to this default.
I found the `--with-target` via:
``` 
$ ../configure --help
...
  --with-target=riscv64-unknown-elf
                          Sets the default target config
```