riscv-software-src/riscv-isa-sim #273: openbsd port errors

hakrdinesh, 2019-02-04:
OpenBSD 6.4/amd64

ob1$ script openbsd.out
Script started, output file is openbsd.out
ob1$ cd riscv-isa-sim                                                                                                                                                   
ob1$ mkdir build
ob1$ cd build
ob1$ export RISCV=/home/dt/r5
ob1$ uname -a
OpenBSD ob1.my.domain 6.4 GENERIC#349 amd64
ob1$ cc --version
OpenBSD clang version 6.0.0 (tags/RELEASE_600/final) (based on LLVM 6.0.0)
Target: amd64-unknown-openbsd6.4
Thread model: posix
InstalledDir: /usr/bin
ob1$ export CC=cc
ob1$ c++ --version
OpenBSD clang version 6.0.0 (tags/RELEASE_600/final) (based on LLVM 6.0.0)
Target: amd64-unknown-openbsd6.4
Thread model: posix
InstalledDir: /usr/bin
ob1$ export CXX=c++
ob1$ ../configure --prefix=$RISV --with-fesvr=$RISCV
ob1$ ../configure --prefix=$RISCV --with-fesvr=$RISCV
checking build system type... x86_64-unknown-openbsd6.4
checking host system type... x86_64-unknown-openbsd6.4
checking for gcc... cc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether cc accepts -g... yes
checking for cc option to accept ISO C89... none needed
checking whether we are using the GNU C++ compiler... yes
checking whether c++ accepts -g... yes
checking for ar... ar
checking for ranlib... ranlib
checking for dtc... /usr/local/bin/dtc
checking how to run the C++ preprocessor... c++ -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking whether byte ordering is bigendian... no
checking for a BSD-compatible install... /usr/bin/install -c
checking for ANSI C header files... (cached) yes
configure: configuring default subproject : riscv
checking for library containing dlopen... none required
checking for libfesvr_is_present in -lfesvr... yes
checking for pthread_create in -lpthread... yes
configure: configuring default subproject : dummy_rocc
configure: configuring default subproject : softfloat
configure: configuring default subproject : spike_main
configure: creating ./config.status
config.status: creating riscv.mk
config.status: creating dummy_rocc.mk
config.status: creating softfloat.mk
config.status: creating spike_main.mk
config.status: creating Makefile
config.status: creating riscv-spike.pc
[... 51 lines cut ...]
  if (setsockopt(socket_fd, SOL_SOCKET, SO_REUSEADDR, &reuseaddr,
                                        ^
../riscv/remote_bitbang.cc:45:22: error: variable has incomplete type 'struct sockaddr_in'
  struct sockaddr_in addr;
                     ^
../riscv/remote_bitbang.cc:45:10: note: forward declaration of 'sockaddr_in'
  struct sockaddr_in addr;
         ^
../riscv/remote_bitbang.cc:47:21: error: use of undeclared identifier 'AF_INET'
  addr.sin_family = AF_INET;
                    ^
../riscv/remote_bitbang.cc:48:26: error: use of undeclared identifier 'INADDR_ANY'
  addr.sin_addr.s_addr = INADDR_ANY;
                         ^
../riscv/remote_bitbang.cc:57:7: error: use of undeclared identifier 'listen'
  if (listen(socket_fd, 1) == -1) {
      ^
../riscv/remote_bitbang.cc:77:17: error: no member named 'accept' in the global namespace
  client_fd = ::accept(socket_fd, NULL, NULL);
              ~~^
9 errors generated.
gmake: *** [Makefile:319: remote_bitbang.o] Error 1
ob1$ ^D
Script done, output file is openbsd.out
ob1$

hakrdinesh, 2019-02-04:
ob1$ gmake
gmake: ../scripts/vcs-version.sh: Command not found

hakrdinesh, 2019-02-04:
Kindly pull https://github.com/hakrdinesh/riscv-isa-sim/commit/5907e70c43c172f456f3bbb52c85ba156d025d55

The error on OpenBSD goes away. No regressions will be seen Ubuntu 16.04, FreeBSD 13.0-current, NetBSD 8.0, and OpenBSD 6.4 on the amd64 platform.

hakrdinesh, 2019-02-04:
Kindly pull https://github.com/hakrdinesh/riscv-isa-sim/commit/ec694bbddad0dc9af38597c7df63f139d67c77ce

to fix compilation errors on OpenBSD. 

The compilation errors are gone with that change. 
ob1$ pwd                                                  
/home/dt/riscv-isa-sim/build
ob1$ gmake 
gmake: Nothing to be done for 'default'.
ob1$ find . -name "*bit*.o"                               
./remote_bitbang.o
ob1$ git remote -v
origin  https://github.com/hakrdinesh/riscv-isa-sim (fetch)
origin  https://github.com/hakrdinesh/riscv-isa-sim (push)
ob1$

Thanks.

hakrdinesh, 2019-02-04:
Need to check for any regressions on NetBSD, FreeBSD and Ubuntu.

FrostKnight, 2019-06-28:
Sorry to bump, but does OpenBSD plan to support Risc-V in the future? And if so, when? just curious.