riscv-software-src/riscv-isa-sim #275: How to find a number of instructions executed for particular program ?

billasurendra, 2019-02-13:
I am executing one executable file on RISCV Linux (Spike Simulator), I want to know how many instructions executed for particular executable.

Thanks In Advance

arsinoe, 2019-04-04:
(Correct me if I'm wrong)
If you see the processor class code in the simulator you will see that there is a _state_ struct. This state (among others) holds the processor's counter registers. If you see now the _execute.cc_ code you will notice that every time an instruction is executed, counter _state.minstret_ increases accordingly. 

_(When we want to read a csr register from an application program we call the read_csr() function that finally ends to the get_csr() routine of the processor class and returns the state struct register.)_ 

So, the only thing you have to do is find the register that corresponds to  _state.minstret_ and read it from your program. In get_csr() we notice that if we read cycle counter it will return the value of  _state.minstret_ register.

Sanoj-S-Vijendra, 2024-06-03:
But how do we know in which register does our state.minsret is stored?