riscv-software-src/riscv-isa-sim #276: Using Spike + OpenOCD + GDB (Could not read registers; remote failure )

fellipe-leandro, 2019-02-18:
Hello, everyone!
I've just started to learn about RISC-V and I'm trying to follow the tutorial provided by the README of this repo. I'm stuck on session **Debugging With Gdb** . I'm using OPENOCD provided by riscv ([https://github.com/riscv/riscv-openocd)](url), and that's my .cfg file: 
 interface remote_bitbang
remote_bitbang_host localhost
remote_bitbang_port 9824

set _CHIPNAME riscv
jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0xdeadbeef

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME riscv -chain-position $_TARGETNAME

gdb_report_data_abort enable

init
#halt

 I have to comment the _halt_ command because with it, the opeocd quits right after it is invoked.  Without halt, this is the screen I get:
 $HOME/myopenocd/bin/openocd -f spike.cfg 
Open On-Chip Debugger 0.10.0+dev-00693-gfd8a3c9 (2019-02-16-23:34)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Warn : Adapter driver 'remote_bitbang' did not declare which transports it allows; assuming legacy JTAG-only
Info : only one transport option; autoselect 'jtag'
Info : Initializing remote_bitbang driver
Info : Connecting to localhost:9824
Info : remote_bitbang driver initialized
Info : This adapter doesn't support configurable speed
Info : JTAG tap: riscv.cpu tap/device found: 0xdeadbeef (mfg: 0x777 (<unknown>), part: 0xeadb, ver: 0xd)
Error: failed read (NOP) at 0x11, status=2
Info : Listening on port 3333 for gdb connections
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333

When I start **gdb** using riscv64-unknown-elf-gdb rot13-64 and then run the command _target remote localhost:3333_, I get this gdb error:
warning: Remote failure reply: E01
Could not read registers; remote failure reply 'E0E'

And then, those messages appears on **openocd** screen:
Error: Target not initialized. Return ERROR_FAIL.
Error: get register list failed
Info : Failed detecting Target Description Support, disabling
undefined debug reason 7 - target needs reset
Error: Target not initialized. Return ERROR_FAIL.
Info : dropped 'gdb' connection

So, what I'm doing wrong that causes this _Could not read registers; remote failure reply 'E0E'_? 
I have compiled the .c source file and run spike just as the README file shows.

timsifive, 2019-02-19:
I just tried this, and the example works just as expected. (I did find one small problem that's past where you're having issues. https://github.com/riscv/riscv-isa-sim/pull/277 will fix that.)

I'm using OpenOCD 8dd5d2a71064e8cab577b6681d08314e919af753, and spike 1916b185e25f9e99f192633eadeac592d64d35c4.

fellipe-leandro, 2019-02-22:
Thanks for the reply, @timsifive !
I've checked the hash versions of both OpenOCD and Spike versions that you used, and they are the same I used.
So, I read a forum issue with a problem quite like mine  [https://forums.sifive.com/t/openocd-jtag-working-incorrectly-with-freedom-soc/1144](url), and, as suggested on it, I changed my linux  version from lubuntu14.04 to lubuntu18.04. I have just ran the example in the README and it worked!!

closed 2019-02-27 by fellipe-leandro

stefanct, 2019-03-08:
@fellipe-leandro that link is more than broken...

fellipe-leandro, 2019-03-08:
Sorry, @stefanct.
I've just update it. It was a Markdown syntax problem (I'm still learning).
Thanks for the notification.

stefanct, 2019-03-08:
@fellipe-leandro The syntax problem was a minor nuisance but I think you meant some other link actually because there is no mention of library problems or ubuntu or any of your error messages. Or do I miss anything?

fellipe-leandro, 2019-03-08:
@stefanct You're right! I I confused the links. Now it is right! Thanks again

paladugushona, 2021-01-24:
i am working on windows
i have the same problem

timsifive, 2021-01-25:
I don't think spike is supported on Windows at all: https://github.com/riscv/riscv-isa-sim/issues/585

WajahatRiaz, 2023-03-01:
> I don't think spike is supported on Windows at all: #585

I believe he means WSL

aswaterman, 2023-03-01:
This thread was initiated and closed four years ago, and was last commented on two years ago.  No need to revive it.