riscv-software-src/riscv-isa-sim #278: yeild_load_reservation dependant on INTERLEAVE value?

mainkar-rohit, 2019-02-21:
I understand that INTERLEAVE value is 5000 as a default for Spike performance. It is not clear to me why we yield the load reservation when current_step is equal to INTERLEAVE.

 `    current_step += steps;
    if (current_step == INTERLEAVE)
    {
      current_step = 0;
      **procs[current_proc]->get_mmu()->yield_load_reservation();**
      if (++current_proc == procs.size()) {
        current_proc = 0;
        clint->increment(INTERLEAVE / INSNS_PER_RTC_TICK);
   `

Am I misunderstanding something?
Wouldn't this return a false fail on SC op if LR is <5000th executed op but SC is a >5000th executed op?

mainkar-rohit, 2019-02-21:
Referencing the code in question for clarity:
https://github.com/riscv/riscv-isa-sim/blob/master/riscv/sim.cc#L111

aswaterman, 2019-02-21:
You are right that it will fail the SC.  I'm not sure what you mean by false pass, but this is valid behavior.

This simple scheme avoids having to invalidate reservations by address. Any store can invalidate another thread's reservation.  Rather than checking the reservations on every store, we unconditionally fail the reservation on thread switches.  As long as threads are switched infrequently enough that a 16-instruction LR/SC loop will eventually succeed, it's a correct implementation of the ISA.

mainkar-rohit, 2019-02-21:
You are correct, I made the correction to avoid confusion.

mainkar-rohit, 2019-02-21:
What do you mean by thread switches in context of Spike?
Is the limit of 16-instruction LR/SC loop coming from the spec?

aswaterman, 2019-02-21:
Spike doesn't run multiple target harts simultaneously.  When simulating multiple harts, it switches which hart it's simulating every 5K instructions.

Yes, the 16-instruction constraint comes from the ISA spec.

mainkar-rohit, 2019-02-21:
Thanks for explaining.
Quoting the spec for posterity:
"The static code for the LR/SC sequence plus the code to retry the sequence in case
of failure must comprise at most 16 integer instructions placed sequentially in memory."

mainkar-rohit, 2019-02-21:
Even for a single hart, Spike simulates in chunks of 5k instructions, right?
In that case, doesn't a corner case exist in Spike wherein SC is >5000th executed instructions and LR is a <5000th executed code, leading to a false failure from SC on the first attempt?

aswaterman, 2019-02-21:
It does result in a failure, but SC is allowed to fail spuriously, so I would not call it a "false" failure.

mainkar-rohit, 2019-02-21:
Thanks for explaining.

closed 2019-02-21 by mainkar-rohit