riscv-software-src/riscv-isa-sim #282: How does spike set breakpoint debugging?What does spike -d pk files output?

mouhuajun, 2019-03-06:
How does spike set breakpoint debugging?What does spike -d pk files output?

aswaterman, 2019-03-06:
`spike -d` is a debug mode, so you can single-step through the instruction stream and examine the architectural state of the machine.  It also provides the ability to run until the PC hits some value, which is effectively a breakpoint.

If you type `help` at the console you see when you run `spike -d program`, you can see some of the very basic commands it allows.

omerguzelelectronicguy, 2024-02-16:
@aswaterman This debug mode is different from what riscv debug specification is right? I mean spike supports debug mode specification of riscv architecture(I guess). but when we run spike with -d flag, it makes it easier to follow the program(not to implement the debug mode of riscv).

aswaterman, 2024-02-16:
That's right.  Spike implements the RISC-V debug architecture, but that has nothing to do with the `-d` command-line argument.

omerguzelelectronicguy, 2024-02-16:
I think this can be closed.

closed 2024-02-16 by aswaterman