riscv-software-src/riscv-isa-sim #286: C.SLLI imm==0 does not trap

ccelio, 2019-03-15:
I apologize if this is a misreading of the ISA manual, but I'm a little confused about the language that says:

>"For RV32C and RV64C, the shift amount must be non-zero."

I don't really  know what "must" means (yes, the assembler barfs on it, but what should spike do?).  

Looking at the source code, `c.slli` never traps (https://github.com/riscv/riscv-isa-sim/blob/master/riscv/insns/c_slli.h). I recognize that "Table 16.3 RVC HINT Instructions" lists `c.slli rd=x0, nzimm!=0` as a HINT, but it does not list `c.slli nzimm==0` as a HINT.

For comparison, the `c.ldsp` throws an exception for `c.ldsp rd=x0` (https://github.com/riscv/riscv-isa-sim/blob/master/riscv/insns/c_lwsp.h), and it's language describes the situation as `is only valid when rd!=0`, but does not list it in the HINT table (I recognize that writing to x0 is not architecturally neutral for loads, so it cannot be used as an architecturally-neutral "HINT").

aswaterman, 2019-03-15:
This isn't entirely obvious, but it is fully specified and Spike's doing the right thing.

In the encoding table, note the C.SLLI64 row.  That instruction has the same encoding as C.SLLI but with imm=0.   It's listed as a HINT for RV32 and RV64.  The HINT table also indicates C.SLLI64 is a HINT for RV32 and RV64.

The reason the text says "the shift amount must be nonzero" is that, if the shift amount is zero, it's a different instruction (C.SLLI64 instead of C.SLLI).

ccelio, 2019-03-15:
Thank you for the clarification. I see that the only `c.slli64` references are in Tables 16.3 and 16.7, but not in the text about `c.slli` so I didn't look at it carefully enough. Perhaps we should add `c.slli64` just below the `c.slli` entry? And I do like Table 16.7 clarifying when a particular instruction is deemed a HINT.  That should've been enough of a giveaway to me.

closed 2019-03-15 by ccelio

aswaterman, 2019-03-15:
If I get around to it, I'll think about how to make the normative text clearer.