riscv-software-src/riscv-isa-sim #288: Interrupt priority among M/S/U privilege mode

AoteJin, 2019-03-28:
In privileged manual, there is no explict definition of priority of interrupts among M/S/U mode.  However, in Spike: https://github.com/riscv/riscv-isa-sim/blob/c01631de5027b211a374a01cbbd046a11926df0e/riscv/processor.cc#L206 , it seems that the least privileged mode, namely U mode, takes highset priority, which contradicts my intuitive understanding. I think the M mode interrupt should be taken with higher priority, since M mode handler is more likely to directly cope with underlying hardare devices, IO etc.

Is there any special consideration to escalte priority of U mode interrupt?

P.S. Given that the PLIC is not neccessarily in place to handle priority. 

Regards & thanks!

aswaterman, 2019-03-28:
Spike follows the convention in the spec (at least it is supposed to...).

The `ctz` is a red herring.  Look at the if/else structure above, which assigns the variable `enabled_interrupts` in priority order, with M-mode taking precedence.

AoteJin, 2019-03-28:
https://github.com/riscv/riscv-isa-sim/blob/c01631de5027b211a374a01cbbd046a11926df0e/riscv/processor.cc#L196
Picturing that MEIP and SEIP happen to be set simultaneously, both pending signals would be sampled by `enabled_interrupt`. Accordingly, the `ctz` seeks for the LSB set to 1 and hereby SEIP is picked. I assume that's not what you expect, right?

aswaterman, 2019-03-28:
I see.  This isn't a problem in practice because SEI is normally delegated to S-mode.  The logic at the top of this function ensures that interrupts are prioritized by destined privilege mode, so when SEI is delegated, the priority is respected.

However, in the situation that SEI is not delegated, this logic would mis-prioritize SEI over MEI.

aswaterman, 2019-03-28:
Fixed, thanks for pointing this out.

closed 2019-03-28 by aswaterman

AoteJin, 2019-03-28:
Thanks for heads-up!

gurjeet, 2019-03-28:
@aswaterman it's customary to reference the commit-id or issue-id in the conversation when the issue leads to a fix or improvement. Can you please reference the commit/change-id here.

aswaterman, 2019-03-28:
Yes, I did.... scroll up or look at this screenshot
[image]

gurjeet, 2019-04-03:
> Yes, I did.... scroll up or look at this screenshot

Oops, sorry I didn't notice the auto-generated message in this conversation.

commit 496c59d0 (2019-03-28): Respect interrupt priorities even when not delegated

The spec says that e.g. MEI takes priority over SEI.  We got this right in
the common case that SEI is delegated to S-mode, but we reversed it in the
undelegated case.

The destination privilege was correct, so this wasn't much of a problem,
but it is technically noncompliant.

Resolves #288