riscv-software-src/riscv-isa-sim #289: How to add an instruction in spike which do exactly what rdcycle do?

ajupazhamayil, 2019-03-28:
I want to add an instruction say rdcount which counts the cycles like rdcycle. Please tell me how to add such an instruction. I already changed the assembler part and successfully it emits my new instruction.

aswaterman, 2019-03-28:
RDCYCLE is actually a special case of of the CSRRS instruction, which reads the CSR named 'cycle'.  You can add more CSRs by extending the switch statement that reads CSR_CYCLE: https://github.com/riscv/riscv-isa-sim/blob/496c59d064961bb81e63e2bba1bdadd4abf05a52/riscv/processor.cc#L622

aswaterman, 2019-03-28:
If you really want to add a new instruction, rather than a new CSR, then you need to

- Create the file in the `riscv/insns` directory
- Add a `DECLARE_INSN` for it in `riscv/encoding.h`
- Add it to `riscv_insn_list` in `riscv/riscv.mk.in`

ajupazhamayil, 2019-03-29:
Thank you @aswaterman. Actually I have done these steps already,
1. In riscv/insns/ added a file rdcount.h
2. generated MATCH_RDCOUNT, MASK_RDCOUNT using riscv-opcodes/opcodes-pseudo
3. in riscv/encoding.h defined match & mask and declared like below
DECLARE_INSN(rdcount, MATCH_RDCOUNT, MASK_RDCOUNT)
4. Added the rdcount in riscv/riscv.mk.in
What I dont understand is,
In rdcount.h I want to write instruction which read the CSR named cycle(I will define new CSR later).
Please tell me how to write that in rdcount.h, Exact instruction/document to my requirement is appreciated.

aswaterman, 2019-03-29:
You can do something like

    WRITE_RD(p->get_csr(CSR_CYCLE));

ajupazhamayil, 2019-03-29:
It is giving me error like this when I run with spike pk a.out
z  0000000000000000 ra 00000000000101c0 sp 000000007f7e9b10 gp 000000000001d0c8
tp 0000000000000000 t0 0000000000000000 t1 0000000000000000 t2 0000000000000000
s0 000000007f7e9b30 s1 0000000000000000 a0 0000000000000001 a1 000000007f7e9b58
a2 0000000000000000 a3 0000000000000001 a4 0000000000000010 a5 0000000000000002
a6 000000000000001f a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 0000000000000000 t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
pc 00000000000101a0 va 00000000c00017f3 insn       c00017f3 sr 8000000200046020
An illegal instruction was executed!

aswaterman, 2019-03-29:
Well, that's because the opcode `c00017f3` is an illegal instruction.  If you try disassembling it, you'll see it's `csrrw a5, cycle, x0`.  Since `cycle` is a read-only CSR, attempting to write it is illegal.

If `c00017f3` is the opcode you tried to assign to your new instruction, you need to pick a different one.  That opcode already belongs to the `csrrw` instruction.

ajupazhamayil, 2019-03-29:
Sorry to ask you simple questions again, I am a beginner here actually,

I got match and mask from riscv-opcodes/opcodes-pseudo via adding the following line and do parse-opcodes -c on that file.
@rdcount    rd 19..15=0 31..20=0xC00 14..12=2 6..2=0x1C 1..0=3

Below is the match and mask I got it to the instruction.
#define MATCH_RDMISS 0xc0001073
#define MASK_RDMISS  0xfffff07f
I presume the instruction opcode is not c00017f3.

And the code Im trying to run is this
```
#include <stdio.h>

static inline uint32_t rdcount(void) {
    uint32_t count;
    asm volatile ("rdcount %0" : "=r"(count));
    return count;
}

int main(){
    uint32_t start;
    start = rdmiss();
    printf("%zu \n", start);
}
```
Can you please suggest me something which I can do to solve the above isssue?

ajupazhamayil, 2019-03-29:
Thank you @aswaterman,
As you said opcode for my new instruction is c00017f3 (c00017f3            rdcount  a5).
I have added line @rdcount    rd 19..15=0 31..20=0xC00 14..12=1 6..2=0x1C 1..0=3 in riscv-opcodes/opcodes-pseudo and got that opcode. Can you please suggest me some document which I can learn about opcode generation for the instruction as I'm a beginner here?

aswaterman, 2019-03-29:
You got it right, except you picked an opcode that belongs to another instruction (`csrrs` in this case).  Pick an opcode that's available, e.g. set bits 6..2=0x1E instead of 0x1C.

`@rdcount rd 19..15=0 31..20=0xC00 14..12=1 6..2=0x1E 1..0=3`

ajupazhamayil, 2019-03-29:
Thank you @aswaterman, After changing that to 1E, It changed to this error.

```
z  0000000000000000 ra 00000000000101c0 sp 000000007f7e9b10 gp 000000000001d0c8
tp 0000000000000000 t0 0000000000000000 t1 0000000000000000 t2 0000000000000000
s0 000000007f7e9b30 s1 0000000000000000 a0 0000000000000001 a1 000000007f7e9b58
a2 0000000000000000 a3 0000000000000001 a4 0000000000000010 a5 0000000000000002
a6 000000000000001f a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 0000000000000000 t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
pc 00000000000101a0 va 00000000c00017fb insn       c00017fb sr 8000000200046020
An illegal instruction was executed!

```
So this should be the problem with opcode c00017fb right? Can you please tell me how to find an opcode which is not given to any instructions?

gurjeet, 2019-03-30:
@ajupazhamayil I believe this question of yours and the problems you faced are a great example of what a newcomer needs to know to get started with a new instruction. Given that RISC-V has official support for custom extensions, anyone needing to develop a new extension would find it helpful.

@aswaterman correct me if I'm wrong.

@ajupazhamayil can you please post complete patch of yours in this issue; perhaps as a code block enclosed in a pair of ```.

ajupazhamayil, 2019-03-30:
@gurjeet Actually I think the following link and the discussion with this issue would describe my steps,
[https://nitish2112.github.io/post/adding-instruction-riscv/](url)

gurjeet, 2019-03-31:
@ajupazhamayil That's great! It's a bit dated, but hopefully the instructions mentioned there still work.

aswaterman, 2019-03-31:
@ajupazhamayil as to how to pick an opcode that is not used by other instructions: examine the ISA spec, Ch. 25, and avoid those.

ajupazhamayil, 2019-03-31:
@aswaterman Thank you very much. It was helpful.

closed 2019-03-31 by ajupazhamayil

chenc6, 2021-05-14:
@ajupazhamayil @aswaterman 
Hi folks,

Do we need to modify other files when we adding a customized instruction for simulation? Do we need to modify **riscv-pk** as well?

I follow a similar process to add my instruction, 
> * Create the file in the `riscv/insns` directory
> * Add a `DECLARE_INSN` for it in `riscv/encoding.h`
> * Add it to `riscv_insn_list` in `riscv/riscv.mk.in`

but unfortunately I still get 
`pc 000000000001016c va/inst 0000000002e787eb sr 8000000200006020
An illegal instruction was executed!`

In debug mode, it's clear to see that the simulator can not identify it.
[image]

The instruction I add is a "mod" R-type instruction, where I use MATCH_MOD 0x200006b, MASK_MOD 0xfe00707f, and DECLARE_INSN(mod, MATCH_MOD, MASK_MOD)

I successfully add it into riscv_gcc, so that in the riscv_objdump file we can see the instruction. I assume there is no conflict with other instructions. 
[image]

In Spike, I define the functionality of mod in mod.h as:
> * WRITE_RD(sext_xlen(RS1 % RS2));

I add it to 'riscv_insn_list' in this way:
[image]

Please let me know where did I make a mistake! I really appreciate it!

Thank you

chihminchao, 2021-05-14:
@chenc6 
Do you fill up the mcounteren or scounteren to enable the access permission ?

chenc6, 2021-05-14:
@chihminchao 
Hi,

Thank you for your reply.

No, I didn't. I am actually not sure what is _mcounteren_ or _scounteren_, and what permission we need. 

Could you please go more specific? Or tell me from which document I can find relevant information?

Thank you.

chihminchao, 2021-05-14:
@chenc6
privilege spec 3.1.12 
"When the CY, TM, IR, or HPMn bit in the mcounteren register is clear, attempts to read the
cycle, time, instret, or hpmcountern register while executing in S-mode or U-mode will cause
an illegal instruction exception. When one of these bits is set, access to the corresponding register
is permitted in the next implemented privilege mode (S-mode if implemented, otherwise U-mode)."