riscv-software-src/riscv-isa-sim #290: Version number

xyproto, 2019-03-29:
spike is now packaged for Arch Linux.

Could you please make a release tag, just to have a version number for the package?

https://www.archlinux.org/packages/community/x86_64/spike/

aswaterman, 2019-03-29:
@palmer-dabbelt can you propose how we should version/release this package and take care of this? See also https://github.com/riscv/riscv-fesvr/issues/65

palmer-dabbelt, 2019-03-30:
Why don't we just call it 1.0.0 and then follow the normal scheme?

gurjeet, 2019-03-30:
@palmer-dabbelt I believe by "normal scheme" you mean [SemVer 2.0][1]. Just curious.

[1]: https://semver.org/spec/v2.0.0.html

sam-falvo, 2019-03-30:
> scheme" you mean SemVer 2.0 <https://semver.org/spec/v2.0.0.html>. Just

Per semver FAQ <https://semver.org/#how-do-i-know-when-to-release-100>, it
seems like you should already be at at least version 1.0.0, so now is as
good a time as any.  (A saying frequently heard in some communities is,
"The best time to plant a tree is 20 years ago.  The second best time is
now.")

> How do I know when to release 1.0.0?
>
> If your software is being used in production, it should probably already
> be 1.0.0. If you have a stable API on which users have come to depend, you
> should be 1.0.0. If you’re worrying a lot about backwards compatibility,
> you should probably already be 1.0.0.
>

aswaterman, 2019-03-30:
This all sounds fine to me, and @sam-falvo's arboreal wisdom rings true here.

aswaterman, 2019-03-30:
https://github.com/riscv/riscv-isa-sim/pull/291

aswaterman, 2019-04-01:
https://github.com/riscv/riscv-isa-sim/releases/tag/v1.0.0

closed 2019-04-01 by aswaterman

xyproto, 2019-04-01:
Updated the Arch Linux package named `spike` to not depend on `fesvr` and to use the `v1.0.0` tag. Thanks!