riscv-software-src/riscv-isa-sim #292: windows+msys2 to Compile and generate spike.exe and pk.exe

topchenhui, 2019-04-02:
We use windows + msys2 to compile and generate riscv-related tool chains: riscv-gnu-tool chain (riscv64-unkown-elf-gcc.exe 64bit), riscv-isa-sim (spike.exe), riscv-pk (pk.exe 64bit). When we tried to run spike PK hello, the command line window had no output.
[image]

aswaterman, 2019-04-02:
Windows is not officially supported, so you are on your own here.  If you figure out what the problem is, please feel free to submit a patch!

Kreijstal, 2024-07-12:
not sure how you compiled it (maybe cygwin)?

but mman.h is not available in windows env
```
g++ -MMD -MP  -DPREFIX=\"/ucrt64\" -Wall -Wno-unused -Wno-nonportable-include-path -g -O2 -fPIC -std=c++17 -g -O2   -I. -I. -I./fesvr -I./riscv -I./disasm -I./customext -I./fdt -I./softfloat -I./spike_main -I./spike_dasm    -c ./fesvr/elfloader.cc
./fesvr/elfloader.cc:10:10: fatal error: sys/mman.h: No such file or directory
   10 | #include <sys/mman.h>
      |          ^~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:349: elfloader.o] Error 1
```

aswaterman, 2024-07-16:
Spike requires a POSIX-compliant system. (`sys/mman.h` is a POSIX header.)

Cygwin is definitely closer than native Win32, but I have my doubts that it'll actually work.  You might have better luck with WSL2.