riscv-software-src/riscv-isa-sim #293: Custom Instruction: How to access non-operand registers

JoshSabet, 2019-04-03:
In the header file that describes a custom instruction (in the riscv/insns directory), how could one access values stored in registers that are not rd, rs1, rs2.

aswaterman, 2019-04-03:
Look at the macro implementations in riscv/decode.h to see how those macros expand. You can write code in-line that looks similar to that.

JoshSabet, 2019-04-05:
If anyone is curious, to write to a register you use the function WRITE_REG(regNumber, value) and to read a register you use the function READ_REG(regNumber) which returns the value at the register number regNumber.

regNumber is the number on the first column of the green card from Patterson and Hennessey

closed 2019-04-05 by JoshSabet