riscv-software-src/riscv-isa-sim #294: Boot Linux on Spike

Jimmyzshs, 2019-04-07:
Hi,

I am trying to boot newest linux kernel with Spike, however it seems to stall after console disabled. I am not sure if any one has the same issue on that:

The setup steps I followed is what I did in 2017 

tool-chain
busybox
build rootfs.cpio
compile linux
compile bbl

spike bbl
bbl loader
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.20.0-44631-g8fe28cb58bcb (yunjie@yunjie-ThinkPad-T430s) (gcc version 8.3.0 (GCC)) #1 SMP Sat Apr 6 23:17:28 CDT 2019
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000fffffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000] software IO TLB: mapped [mem 0xfa3fe000-0xfe3fe000] (64MB)
[    0.000000] elf_hwcap is 0x112d
[    0.000000] percpu: Embedded 17 pages/cpu @(____ptrval____) s29464 r8192 d31976 u69632
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 516615
[    0.000000] Kernel command line: 
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 1988356K/2095104K available (5505K kernel code, 331K rwdata, 1760K rodata, 576K init, 806K bss, 106748K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=1.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.000000] Console: colour dummy device 80x25
[    0.000000] printk: console [tty0] enabled
[    0.000000] printk: bootconsole [early0] disabled

Thanks!

matthiakl, 2019-05-05:
Hi,
I have the same problem trying to load the latest LTS kernel (4.19.37). Even though an initramfs is linked into the kernel, the bootloader bbl is not able to find and load it. The same kernel works fine, when booting using qemu. I already tried different toolchains: riscv64-unknown-elf (7.2.0), riscv64-unknown-linux-gnu (7.2.0), riscv64-linux-gnu (8.3.0).
Any ideas would be helpful, thanks.

matthiakl, 2019-05-17:
@Jimmyzshs 
Try the versions tagged in the v1_0 branch of freedom-u-sdk (https://github.com/sifive/freedom-u-sdk/tree/v1_0). Using the kernel and toolchain from this repo I got it working.