riscv-software-src/riscv-isa-sim #296: Spike headers are no longer installed

ben-marshall, 2019-04-18:
I see that in [this commit](https://github.com/riscv/riscv-isa-sim/commit/f49618ca9d674ec7e596118f85027c4b503862ac) the installation of spike headers was removed. Now only the fesvr headers are installed.

What was the rationale for this? It's broken a bunch of stuff for me (unimportant in the grand scheme, I know) and seemed like a useful thing for people building on spike as part of other projects?

aswaterman, 2019-04-18:
If you want to open a PR that changes the installation behavior to install the headers again, I’d merge it.

ben-marshall, 2019-04-18:
Okay, thanks. Is there anything you would want such a merge to avoid? I don't want to include something in an installation which has been left out by design.

aswaterman, 2019-04-18:
Well, as the readme says, the C++ interfaces aren't considered a stable API, so relying on any of the headers involves some degree of risk.  With that caveat, installing all of them seems OK to me.  If we later find out that was a mistake, we can retract some of them without breaking any promises...