riscv-software-src/riscv-isa-sim #300: spike can't run riscv64-unknown-linux-gnu compiled executable file

ericwu13, 2019-05-12:
I want to use Spike to simulate the glibc compiled program since newlibs version can't add -pg option to do gprof. I want to use riscv64-unknown-linux-gnu-gcc to enable -pg option. But I stuck just in the beginning.

I compiled my program using below command.
```bash
$ riscv64-unknown-linux-gnu-gcc hello.c 
```
And I use riscv-tools to build spike and pk, setting host of build-spike-pk.sh to riscv-unknown-linux-gnu
and successfully build them

Later I run the below command

```bash
$ spike pk a.out
```
It turn out segfault (my pk is pointing to the riscv-unknown-linux-gnu/bin/pk). The below is error message.
[image]

My program
```c++
#include <stdio.h>
int main() {
     printf("Hello\n");
     return 0;
}
```
I wonder if spike can simulate the glibc, and how to do it? Or should I use qemu to do? If so, can anyone provide the link of riscv qemu installation (the github is archived though...). Thanks!

aswaterman, 2019-05-12:
Spike can only run statically linked programs, whereas the Linux compiler dynamically links by default. If you add the -static flag it will work.

ericwu13, 2019-05-13:
I've added static flag to the compiler, however, it printed the below error

[image]

I used ct-ng to build the riscv64-unknown-linux-gnu toolchains, with kernel version 4.19, how to solve it? I encountered the same problem running qemu-riscv64, too.

Another question is, I need to do this to enable pk
```
$ spike $(which pk) <BIN>
```
instead of the below, which will result in the next error, I've set isa-sim target and pk host to riscv64-unknown-linux-gnu, but still can't work. And PATH is properly set ($RISCV/bin and $RISCV/riscv64-unknown-linux-gny/bin)
```
$ spike pk <BIN>
```
[image]

aswaterman, 2019-05-13:
Ah, my Linux compiler must be older than yours. In that case, I recommend
using an elf compiler instead of a linux one.

ericwu13, 2019-05-13:
but the elf one can't add -pg flag to do profiling. I think I'll try build an older linux compiler.

ericwu13, 2019-05-13:
I've solved the problem by installing riscv compiler using riscv-gnu-toolchains, instead of ct-ng. I think it is because my gcc and c version is 5.4 while i'm using 8.3 in ct-ng... But I can't change it to lower version though. Anyone, using riscv-gnu-toolchains works for me.

closed 2019-05-13 by ericwu13