riscv-software-src/riscv-isa-sim #301: PATCH: Don't need to set MSTATUS_UXL/SXL for RV32.

mintow, 2019-05-23:
For RV32, mstatus will be set to 0x5 0000 0000 in processor_t::reset().
It's not reasonable. 
Refer to 3.1.8, Privileged Architecture Version 1.10:
For RV32 systems, the SXL and UXL fields do not exist.

And I found two lines of duplicate ‘MSTATUS_UXL’ code, the patch deleted one.

``` c++
From 3abc8ac4f13ef0f86bb9f7ddb7155f427f55d107 Mon Sep 17 00:00:00 2001
From: matteo <duanmintao@allwinnertech.com>
Date: Thu, 23 May 2019 17:26:49 +0800
Subject: Don't need to set MSTATUS_UXL/SXL for RV32.

Refer to 3.1.8, Privileged Architecture Version 1.10:
For RV32 systems, the SXL and UXL fields do not exist.

---
 riscv/processor.cc | 10 +++++-----
 1 file changed, 5 insertions(+), 5 deletions(-)

diff --git a/riscv/processor.cc b/riscv/processor.cc
index 189b7db..ee906ed 100644
--- a/riscv/processor.cc
+++ b/riscv/processor.cc
@@ -397,13 +397,13 @@ void processor_t::set_csr(int which, reg_t val)
       dirty |= (state.mstatus & MSTATUS_XS) == MSTATUS_XS;
       if (max_xlen == 32)
         state.mstatus = set_field(state.mstatus, MSTATUS32_SD, dirty);
-      else
+      else {
         state.mstatus = set_field(state.mstatus, MSTATUS64_SD, dirty);
+        state.mstatus = set_field(state.mstatus, MSTATUS_UXL, xlen_to_uxl(max_xlen));
+        state.mstatus = set_field(state.mstatus, MSTATUS_SXL, xlen_to_uxl(max_xlen));
+        // U-XLEN == S-XLEN == M-XLEN
+      }
 
-      state.mstatus = set_field(state.mstatus, MSTATUS_UXL, xlen_to_uxl(max_xlen));
-      state.mstatus = set_field(state.mstatus, MSTATUS_UXL, xlen_to_uxl(max_xlen));
-      state.mstatus = set_field(state.mstatus, MSTATUS_SXL, xlen_to_uxl(max_xlen));
-      // U-XLEN == S-XLEN == M-XLEN
       xlen = max_xlen;
       break;
     }
-- 
2.17.1
```

aswaterman, 2019-05-23:
> Signed-off-by: matteo <duanmintao@allwinnertech.com>

I don’t think this is actually a bug. When software attempts to read the
CSR, the MSBs will be zapped, and the value written to rd will not contain
UXL and SXL. This is because of the sext_xlen macro invocation:
https://github.com/riscv/riscv-isa-sim/blob/master/riscv/insns/csrrc.h#L7

In other words, the state.mstatus implementation is partially
microarchitectural, and as long as those fields can’t become
architecturally visible, they don’t matter.

mintow, 2019-05-24:
> I don’t think this is actually a bug. When software attempts to read the CSR, the MSBs will be zapped, and the value written to rd will not contain UXL and SXL. This is because of the sext_xlen [...]

Thank you for explaining！
Yes，it does not affect the function of spike.
Maybe I can say that it is not enough logical enough. :)

aswaterman, 2019-05-24:
That is a fair complaint. The code is seriously lacking in comments. Sorry about that.

MDBrothers, 2019-08-23:
@aswaterman
Why does line 504 duplicate line 503?
CODE: state.mstatus = set_field(state.mstatus, MSTATUS_UXL, xlen_to_uxl(max_xlen));

aswaterman, 2019-08-23:
Hmm, looks like I messed up.  I'll fix that.  Fortunately the duplicated statement has no effect.

MDBrothers, 2019-08-23:
Thanks for taking a look at it.