riscv-software-src/riscv-isa-sim #302: how to add a device in spike?

weiqi7777, 2019-05-24:
hi , i add one device uart in spike, but it can't work.

below is the uart.h file.
`
#define UART_SIZE  0x1000

class uart_t : public abstract_device_t {
 public:
   uart_t() {
      std::cout << "create uart component" << std::endl;
   }
   ~uart_t() {}
    bool load(reg_t addr, size_t len, uint8_t* bytes);
    bool store(reg_t addr, size_t len, const uint8_t* bytes);
    size_t size() {return UART_SIZE;}

 private:
    uint32_t control_register=8;
    uint32_t reserve_register=7;
    uint8_t data_register=5;
    std::string s_data;
}
`

and i add below code in sim_t::sim_t in sim.cc
`
#define UART_BASE 0x50000000
std::unique_ptr<uart_t> uart;
uart.reset(new uart_t);
bus.add_device(reg_t(UART_BASE), uart.get());
`
when i execute  sb  ins, store a byte to 0x5000000, and i get "exceptio trap_store_access_fault".

how can i do to add the device into spike?

weiqi7777, 2019-06-14:
it's my code fault.
the code should be 
`
#define UART_BASE 0x50000000  
uart_t *uart;
uart = new uart_t();
bus.add_device(reg_t(UART_BASE), uart);
`

closed 2019-06-14 by weiqi7777

alejandrolabajos, 2024-10-16:
Hello @weiqi7777, I am struggling through some problems trying to find detailed documentation for defining memory mapped devices in Spike. Could you provide some guidance about what files need to be modified and how to run a test code?