riscv-software-src/riscv-isa-sim #305: specification's of spike.

cghalibkhan, 2019-06-14:
I want to know the following details on the spike.

1>  What is the maximum application size spike can process?
For instance:
I tried with
$ spike pk hello 
$ spike pk bbl
$ spike pk vmlinux
And tried with other applications of riscv but stills
Here vmlinux is not working, it's failed.

can Anyone Help me to find out the limitations of spike that, How large application does it process, if know tell in a detailed manner or if u have doc's, site link's, are else details regarding spike, Let me know.

2> I want to get and study trace from spike for multicore application.
   For that I Run, the following command,
$  riscv64-unknown-elf-gcc hello.c -o hello
$  spike -l -p(n) pk hello > trace-log
i.e; n -- Number of cores.

Here Got the trace, but can't able to study it.

--> Are else Spike does have an option -p(n). Is spike good to consider for this? 

3> Is there any simulator other than spike & qemu, to get Trace of instructions.
      If there Let me know.

4> Are there any other alternatives to get a trace of multicore application?

Thanks,
khan.

aswaterman, 2019-06-14:
Spike can handle arbitrarily large applications, if you give it enough memory (`-m` option).  Note that bbl/vmlinux aren't meant to be arguments to `pk`.  You should just do `spike bbl`.

cghalibkhan, 2019-06-14:
if i Do
$ spike bbl
ERROR: invalid load from debug module: 8 bytes at 0x000000000001dbc8
terminate called after throwing an instance of 'trap_load_access_fault'

spike is working with pk only, otherwise getting same Error for simple Hello world code.

sathya281, 2020-02-14:
usually bbl is generated with proxy kernel. you need not give the pk argument for bbl images.