riscv-software-src/riscv-isa-sim #306: I have builld success the rvv Spike and binutils rvv 0.7.1 and run on gcc inline assembly. It is look very good. I will try some algorithm on this platform. Thank you.

myftptoyman, 2019-06-16:
I have some problem on vslide1up_vx or vslide1down_vx
vslide1up.vx vd, vs2, rs1, vm        # vd[0]=x[rs1], vd[i+1] = vs2[i]
if vd==vs2, the vslide1up will copy all vd[i] as x[rs1]

I expect the value is 
  vslide1up.vx v1, v1, x1 =>
 for(i=0; i<(vl-1); i++) {
   v1[i+1] = v1[i]   
 }
   v1[0] = x1, 

But simulation is  
 v1[0] = x1
 for(i=0; i<(vl-1); i++) {
   v1[i+1] = v1[i]   
 }

> I have builld success the rvv Spike and binutils rvv 0.7.1 and run on gcc inline assembly. It is look very good. I will try some algorithm on this platform. Thank you.

aswaterman, 2019-06-16:
@chihminchao can you look into this? I think slideup/slide1up need to iterate from vl-1 to 0, rather than iterating from 0 to vl-1.

chihminchao, 2019-06-17:
It is a bug and will be fix in next force-push of my PR.

chihminchao, 2019-06-17:
Just make sure one thing
By the spec [17.4.3](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#1743-vector-slide1up)
"The vslide1up instruction requires that the destination vector register group does not overlap the source vector register group and the mask register if masked. Otherwise, an illegal instruction exception is raised."

By the spec [17.4.4](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#1744-vector-slide1down-instruction)
"The vslide1down instruction requires that the destination vector register group does not overlap the mask register if masked. Otherwise, an illegal instruction exception is raised."

In other words,
"vslide1up" allow vd overlap vs2 but "vslide1down" don't
The overlapping definitions differ between these two instructions.  In present implementation, we will allow overlapping and fix it in next push.

chihminchao, 2019-06-17:
@myftptoyman  The conclusion is not to allow vd == vs2 for vslide1up after [discussion](riscv/riscv-v-spec#214) about the spec.  But we still forgot to check the overlapping condition and trigger illegal instruction exception.  I will add this check in next PR.

myftptoyman, 2019-06-18:
I have another problem is the opcode generate by risc/opcode like different current version in Isa simulator？

davetw, 2019-06-19:
hi @myftptoyman could you point out where is the difference?

chihminchao, 2019-06-19:
@myftptoyman the latest change is still under reviewing in PR https://github.com/riscv/riscv-opcodes/pull/29