riscv-software-src/riscv-isa-sim #312: C_JAL and C_ADDIW decoding issue

taoliug, 2019-07-19:
It seems to me C_JAL and C_ADDIW have the same encoding in spike

C_JAL
https://github.com/riscv/riscv-isa-sim/blob/master/riscv/encoding.h#L673
#define MATCH_C_JAL 0x2001
#define MASK_C_JAL  0xe003

C_ADDIW
https://github.com/riscv/riscv-isa-sim/blob/master/riscv/encoding.h#L643
#define MATCH_C_ADDIW 0x2001
#define MASK_C_ADDIW  0xe003

Here's the spike simulation log:
core   0: 0xffffffff8000f31c (0x0000302d) addiw   zero, zero, -21

0x302d is decoded as c_addiw instruction. However, when bit[11:7] !=0, it should be decoded as C_JAL instruction? Alternatively, if rd == 0 for C_ADDIW instruction, it should be decoded as a reserved instruction?

[image]

aswaterman, 2019-07-19:
C.JAL is an RV32-only instruction and C.ADDIW is an RV64-only instruction, so although they share the same opcode, the meaning is unambiguous.

taoliug, 2019-07-19:
In this case, should 0x302d be decoded as illegal C.ADDIW instruction as rd == 0?

aswaterman, 2019-07-19:
Yes, in RV64 mode, 0x302d is an illegal C.ADDIW with rd = 0.

It looks like versions of spike before 2016 did not trap when rd = 0, but instead treated the instruction as a no-op.  So if it is not trapping, I suggest upgrading.

taoliug, 2019-07-19:
BTW, I am simulating with --isa=rv32imc, I think it should be decoded as C_JAL in this case.

aswaterman, 2019-07-19:
Yes.  When I tried that, it disassembled correctly.  It's possible that was a disassembler bug that we have fixed in a more recent version than you are using.

taoliug, 2019-07-19:
Just installed the latest version, the issue is gone, thanks for your help!

closed 2019-07-19 by taoliug