riscv-software-src/riscv-isa-sim #317: Version of host compiler required

eroom1966, 2019-07-29:
Which gcc version of the host compiler is required to build riscv-isa-sim ?
I am seeing the following errors, I have tried switching between gcc 7,8 & 9 but all produce similar errors

> g++ -fPIC -MMD -MP -I/scratch/install/20190729/include  -Wall -Wno-unused -g -O2 -std=c++11 -DPREFIX=\"/scratch/install/20190729\" -I. -I.. -I../riscv -I../dummy_rocc -I../softfloat -I../spike_main -x c++-header ../riscv/insn_template.h -o insn_template.h.gch
> In file included from ../riscv/mmu.h:6:0,
>                  from ../riscv/insn_template.h:3:
> ../riscv/decode.h: In function â€˜velt_t velt(float16_t)â€™:
> ../riscv/decode.h:577:44: error: expected primary-expression before â€˜.â€™ token
>  inline velt_t velt(float16_t f) { return { .h[0]=f.v }; }

aswaterman, 2019-07-29:
The rvv branch is stale and is not actively supported; I recommend using the master branch, which has experimental support for a more recent version of the v extension.

eroom1966, 2019-07-29:
@aswaterman 
thanks for that, I thought the Master did not have the vectors.
I have switched to the Master and all is fine.
Thx

closed 2019-07-29 by eroom1966