riscv-software-src/riscv-isa-sim #329: Error when following "Debugging with Gdb" section in Readme.md

luoxinjie1146, 2019-09-06:
When I run `spike --rbb-port=9824 -m0x10000000:0x20000 rot13-64`, the following error occured:
```
Listening for remote bitbang connection on port 9824.
warning: tohost and fromhost symbols not in ELF; can't communicate with target
```
Is this correct or I should do something to fix it?

timsifive, 2019-09-06:
The output of spike looks normal.

The output of OpenOCD indicates you're running an OpenOCD that doesn't have RISC-V support. You can get one that does at https://github.com/riscv/riscv-openocd.

luoxinjie1146, 2019-09-07:
I tried to build OpenOCD from [https://github.com/riscv/riscv-tools](url), should I add some options when running `./configure`?

timsifive, 2019-09-09:
You need `--enable-remote-bitbang` to be able to talk to spike.

luoxinjie1146, 2019-09-10:
I used the following commands,
```
--prefix=$RISCV --enable-remote-bitbang --enable-jtag_vpi --disable-werror
make
make install
```
Did I miss something?

timsifive, 2019-09-10:
That looks good to me.

luoxinjie1146, 2019-09-12:
Thanks a lot. I didn't running the right openocd binary! I will close this issue now

closed 2019-09-12 by luoxinjie1146

WilliamWangPeng, 2019-09-29:
@timsifive  
hi Tim at SiFive
    Nice to meet you, dear Tm at SiFive, This is William, I have some questions about debugging with GDB following the website
https://github.com/riscv/riscv-isa-sim
I followed the steps on debugging with GDB, inputting the parameters
https://github.com/riscv/riscv-isa-sim/issues/329

--prefix=$RISCV --enable-remote-bitbang --enable-jtag_vpi --disable-werrormake 
make install

  but getting some results below:

pwang@york:~/test$ openocd -f spike.cfg
Open On-Chip Debugger 0.10.0+dev-00198-g35eed36ff (2019-09-29-23:43)
Licensed under GNU GPL v2
For bug reports, read
    http://openocd.org/doc/doxygen/bugs.html
Warn : Adapter driver 'remote_bitbang' did not declare which transports it allows; assuming legacy JTAG-only
Info : only one transport option; autoselect 'jtag'
Info : Initializing remote_bitbang driver
Info : Connecting to localhost:9824
Error: Failed to connect: Connection refused

do you know what is wrong and how can I fix it?
 thank you so much
Yours William

timsifive, 2019-09-30:
It looks like spike isn't running, or you didn't use --rbb-port=9824 when
you started it.

Tim