riscv-software-src/riscv-isa-sim #332: Write-able field in CSR SIP

ma142857, 2019-09-16:
I have a question regarding which bits in the SIP CSR is write-able. According to the Spike implementation, only Supervisor Software Interrupt Pending could be set for the SIP CSR. 

https://github.com/riscv/riscv-isa-sim/blob/6d15c93fd75db322981fe58ea1db13035e0f7add/riscv/processor.cc#L559-L562

However, the privileged spec stats that 
> If an interrupt is delegated to privilege mode x by setting a bit in the mideleg register, it becomes visible in the xip register and is maskable using the xie register. 

And 

> A supervisor-level software interrupt is triggered on the current hart by writing 1 to its supervisor software interrupt-pending  (SSIP)  bit  in  the sip register.

So "reg_t mask = MIP_SSIP | state.mideleg;" makes more sense to me. 
Can you confirm this is a typo or Spike only supports SSIP to be writable in SIP intentionally? 

Best,
Andy

aswaterman, 2019-09-16:
It's intentional.  In most systems, SEIP is wired in directly from an interrupt controller (though Spike doesn't have one).  STIP is managed by M-mode software in current systems, and keeping it read-only from S-mode will make it easier to add an extension that adds a dedicated S-mode timer.

ma142857, 2019-09-16:
I see. So the "mask" here should be a sub-set of state.mideleg.
Thanks for the quick response.