riscv-software-src/riscv-isa-sim #333: instruction count in log

kalairajah, 2019-09-23:
Hi,
 Is there a way to print the instruction number before each instruction when spike executes it, Currently, I don't see any option that can be passed to spike to print this information. 

Here is an example of what I currently see in the logs for an instruction executed ..

core 0:  0x00000000800001664 (0xc9d80d93) addi s11, a6, -867
3 0x00000000800001664 (0xc9d80d93) x27 0xffffffffe8866c9c

I donot use pk and run spike as follows:
spike -l elf 2>& spike.out

Thanks,
Kalai

aswaterman, 2019-09-23:
You can post-process the log using something like `nl spike.out`

kalairajah, 2019-09-24:
Thanks . Is there any plan to add it as part of the regular run instead of post processing the output.

aswaterman, 2019-09-24:
No, it's not practical to add command-line options to make the log look exactly right for each use case, so we'd prefer to rely on post-processing to clean things up.

closed 2019-09-24 by aswaterman