riscv-software-src/riscv-isa-sim #338: Question: Should Spike always stop when it sees an ecall instruction?

sdexter-numa, 2019-10-01:
I'm seeing that there are some cases where Spike does not stop it's simulation when it sees an ecall instruction. It causes problems, because it throws an exception when this happens, and continues executing code that is not meant for Spike.

Is it a bug with the simulator if it won't stop when it sees ecall?

aswaterman, 2019-10-01:
ECALL is just another instruction, which happens to cause an exception.
There’s no reason the simulator should stop executing upon ECALL.

sdexter-numa, 2019-10-01:
Thanks for your response. Interesting that the simulator would usually stop after decoding an ecall instruction. Is that unexpected behaviour?
Follow up question: Is there any surefire way of triggering a Spike simulation stop?

aswaterman, 2019-10-01:
There’s a memory-mapped host-target interface that can be used, among other
things, to end the simulation. This is how the riscv-tests report
pass/failure.

sdexter-numa, 2019-10-03:
Thanks, I was able to identify this in my tests and resolve the issue.

closed 2019-10-03 by sdexter-numa