riscv-software-src/riscv-isa-sim #339: Widening vector arithmetic instructions not checking for correctness of destination vector register group as per the LMUL value

deepsrc, 2019-10-01:
The vector specification states that _"the destination vector register group must be specified using a vector register number that is valid for the destination’s LMUL value, otherwise an illegal instruction exception is raised."_

I have a test which is running with the VLMUL set to 1. So any widening operation with an odd numbered vector register number is supposed to generate an illegal instruction trap. But in case of Spike the instruction executes successfully. A snippet from the failing test is given below -

> core   0: 0xffffffc4cec01542 (0x87fbafd7) vdiv.vv v31, v31, v23
> **core   0: 0xffffffc4cec01546 (0xcec621d7) vwsub.vv v3, v12, v12**
> core   0: 0xffffffc4cec0154a (0xc0202353) fcvt.l.s t1, ft0
> core   0: 0xffffffc4cec0154e (0x8c0d3c83) ld      s9, -1856(s10)

Similar issue exists with the narrowing operations as well where the checks are not applied on the source register numbers.

@khannaudit

luoxinjie1146, 2019-10-02:
Hi @deepsrc , I tried to run a vector test but I didn't found a compilor in riscv-gnu-toolchain. Could you please show where your compilor comes from?

deepsrc, 2019-10-03:
Hi @luoxinjie1146, I think the rvv-0.7.1 branch of the RISC-V GNU toolchain has the support for vector instructions.

deepsrc, 2019-10-03:
@davetw

luoxinjie1146, 2019-10-03:
Thank you @deepsrc very much! I'll try it later.

chihminchao, 2019-10-07:
@deepsrc  It is a bug.  I am  working on it.
Thanks for you report

chihminchao, 2019-11-18:
@deepsrc 
The register usage checker has been merged into master and all illegal register access will trigger illegal instruction exception.