riscv-software-src/riscv-isa-sim #341: Question: vredmax.vs instruction - Documentation for this?

sdexter-numa, 2019-10-03:
I'm using Google's riscv-dv to generate assembly code for Spike, and in a test where illegal instructions are sent, one such instruction was processed by Spike as valid:

core   0: 0xffffffff8000a020 (0x1e4c2857) vredmax.vs v16, v4, v24
3 0x8000a020 (0x1e4c2857)

I've no idea what this vredmax.vs instruction is, and why Spike doesn't detect this as an illegal instruction in ISA rv32IM. I've searched the RISCV documentation for this instruction and cannot find anything.

Could someone direct me to where I can read more about this? Why would Spike pick this up as a valid instruction?

sdexter-numa, 2019-10-03:
I understand it to be a vector instruction, but can't see why Spike would see it as valid if it's not part of the standard ISA, and it's running with ISA set to rv32IM

chihminchao, 2019-10-14:
vector requirement checking doesn't behavior correctly and will be fixed in future update.

chihminchao, 2019-11-18:
@sdexter-numa 

The issue has been fixed in master.