riscv-software-src/riscv-isa-sim #342: Enabling commit log does not work

abdullahyildiz, 2019-10-04:
When I try to configure and then build Spike (using latest commit 9443c1dbac0301faf3a47c5e6914cc7dcb34983e), I couldn't get a log which includes processor register updates:

This is what I did to configure and build Spike
../configure  --enable-commitlog --with-isa=RV64IMAFD --prefix=$RISCV
make

By the way, there is no such a problem for commit 88a852836acb4c7166b1aa4102e11354bfd99234.

amitgargmentor, 2019-10-04:
Hi, 

i actually have one question related to installation:
when you run the ../configure  --enable-commitlog --with-isa=RV64IMAFD --prefix=$RISCV
 command, do you see the below issue :

[amgarg@soc-pe-001 build]$ ../configure --enable-commitlog --with-isa-RV64IMAFD --prefix=$RISCV
configure: WARNING: unrecognized options: --with-isa-RV64IMAFD
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for ar... ar
checking for ranlib... ranlib
checking for dtc... no
**configure: error: device-tree-compiler not found**

abdullahyildiz, 2019-10-05:
No, I can successfully configure, build and then run Spike.

amitgargmentor, 2019-10-07:
ok, Thanks!

aswaterman, 2019-10-07:
It's working fine for me on the head of master.  Just to make sure, you passed the `-l` flag, right?

abdullahyildiz, 2019-10-07:
The following line is an example of how I run Spike for two different builds of it:

`./spike --isa=rv64g -l ../../../riscv-dv/out_2019-10-04/asm_tests/riscv_arithmetic_basic_test_0.o`

aswaterman, 2019-10-07:
Yeah, `./spike --isa=rv64g -l myprogram` works fine for me:

    $ ./spike --isa=rv64g -l pk 2>&1 |head
    core   0: 0x0000000000001000 (0x00000297) auipc   t0, 0x0
    core   0: 0x0000000000001004 (0x02028593) addi    a1, t0, 32

abdullahyildiz, 2019-10-07:
I get a different output for commit https://github.com/riscv/riscv-isa-sim/commit/88a852836acb4c7166b1aa4102e11354bfd99234:

```
cat riscv_arithmetic_basic_test.0.log | head
core   0: 0x0000000000001000 (0x00000297) auipc   t0, 0x0
3 0x0000000000001000 (0x00000297) x 5 0x0000000000001000
core   0: 0x0000000000001004 (0x02028593) addi    a1, t0, 32
3 0x0000000000001004 (0x02028593) x11 0x0000000000001020
core   0: 0x0000000000001008 (0xf1402573) csrr    a0, mhartid
3 0x0000000000001008 (0xf1402573) x10 0x0000000000000000
core   0: 0x000000000000100c (0x0182b283) ld      t0, 24(t0)
3 0x000000000000100c (0x0182b283) x 5 0x0000000080000000
core   0: 0x0000000000001010 (0x00028067) jr      t0
3 0x0000000000001010 (0x00028067)
```

aswaterman, 2019-10-07:
Thank for pasting the output, now I see the issue.  In commit c171379c7828ae94d969846874a7ac542dbda2c3 we made the commit log a command-line option rather than making it always enabled when configured on.  You need to pass `--log-commits`.  I have updated the changelog to reflect this change.  Sorry for the confusion.

abdullahyildiz, 2019-10-07:
OK. Thanks for your help.