riscv-software-src/riscv-isa-sim #343: problem in installation

amitgargmentor, 2019-10-04:
Hi, 

README file mention the below steps to install the SPIKE ISA SIM. 
 apt-get install device-tree-compiler

since i have RED HAT OS, i use the below command:
sudo tum install device-tree-compiler 

but it gives me below error:

**Loaded plugins: langpacks, product-id, search-disabled-repos, subscription-manager
This system is not registered with an entitlement server. You can use subscription-manager to register.
No package device-tree-compiler available.
Error: Nothing to do**

can someone please help ? 

Thanks

chihminchao, 2019-10-14:
I think the package in RedHat distribution may differ from Ubuntu.
Have you tried `yum install dtc`

amitgargmentor, 2019-10-14:
the issue is already resolved. I need to install DTC.

closed 2019-10-14 by amitgargmentor