riscv-software-src/riscv-isa-sim #344: spike trace log not dumping register value

amitgargmentor, 2019-10-07:
Hi, I am also facing the issue of not getting core register value dumped in log file even while configuring the spike I used --enable-commitlog 
../configure --enable-commitlog --prefix=$RISCV

I ran the below command: 
./bin/spike --isa=rv64gc -l riscv_arithmetic_basic_test_1.o  &> riscv_arithmetic_basic_test.1.log

The output is like below:
core   0: 0x0000000000001000 (0x00000297) auipc   t0, 0x0
core   0: 0x0000000000001004 (0x02028593) addi    a1, t0, 32
core   0: 0x0000000000001008 (0xf1402573) csrr    a0, mhartid
core   0: 0x000000000000100c (0x0182b283) ld      t0, 24(t0)
core   0: 0x0000000000001010 (0x00028067) jr      t0
core   0: 0x0000000080000000 (0x00000013) nop
core   0: 0x0000000080000004 (0x0003c0b7) lui     ra, 0x3c
core   0: 0x0000000080000008 (0x3530809b) addiw   ra, ra, 851
core   0: 0x000000008000000c (0x00e09093) slli    ra, ra, 14
core   0: 0x0000000080000010 (0xef808093) addi    ra, ra, -264
core   0: 0x0000000080000014 (0x00004131) c.li    sp, 12
core   0: 0x0000000080000016 (0x0010019b) addiw   gp, zero, 1
core   0: 0x000000008000001a (0x01f19193) slli    gp, gp, 31
core   0: 0x000000008000001e (0x000f8237) lui     tp, 0xf8
core   0: 0x0000000080000022 (0x92d2021b) addiw   tp, tp, -1747

Thanks,
Amit

aswaterman, 2019-10-07:
Add the --log-commits option to the command line when you run Spike

amitgargmentor, 2019-10-07:
Thanks , it worked.

closed 2019-10-07 by amitgargmentor

chihminchao, 2022-06-01:
@Manahil14 
The autotool option is "--enable-commitlog".
The spike running option is "--log-commits"

Manahil14, 2022-06-01:
Beg your pardon, I am still unable to get.

Manahil14, 2022-06-01:
@chihminchao Got it, thanks

akashtenkale03, 2023-01-09:
--log-commits worked thanks.
the command is to determine d cache miss (d_cache.riscv >> executable name)
spike --dc=256:4:32 --log-cache-miss -l --log-commits d_cache.riscv 2>&1 | tee file1.txt

thewesley77, 2023-01-20:
Has there been an update. I cannot get the '--enable-commitlog' to work when using the configure call. I get 'configure: WARNING: unrecognized options'

Is there a new way to enable register changes and have spike output it?

jerryz123, 2023-01-20:
This was recently changed by #1189.

Now you just pass the `--log-commits` to `spike`. The `--enable-commitlog` configure flag is not necessary.

thewesley77, 2023-01-20:
Thanks, didn't even try it with the enable-commitlog.