riscv-software-src/riscv-isa-sim #346: Segmentation Fault on vlw_v instruction

Chert-Nik, 2019-10-11:
Spike commit bbe881f3c5435d02eeb7c28515bfb301470f2875

Segmentation Fault happens during execution of this test generated by [MicroTESK](https://forge.ispras.ru/projects/microtesk):
https://github.com/ispras/riscv-avs/tree/master/testsuite/synthetics/rv32v

Execution of `vlw.v   v30, (t1)` instruction goes as follows:
1. In file **vlw_v.h** macro VI_LD  is executed
2. In file **decode.h** in macro VI_LD line 1454 is executed
`P.VU.elt<uint32_t>(vd + fn * vlmul, vreg_inx) = is_valid ? val : 0;`
actual parameters of the P.VU.elt call will be P.VU.elt<uint32_t>(30, 12);
3. In file **processor.h** in function vectorUnit_t::elt line 182 is executed with vReg == 33; (there is only 32 vector registers!)
4. Value returned by elt function will be &(vectorUnit_t::reg_file[528]).
5. But vectorUnit_t::reg_file was allocated at processor.cc:204 with size of 512 bytes
`reg_file = malloc(NVPR * (VLEN/8));`
6. Some of this wrong memory accesses modify pointer to clint
7. During execution of `ecall` instruction everything segfaults at clint_t::increment() call

Looks like some mistake in vector register indexing.

Test compilation:
riscv64-unknown-elf-gcc -march=rv64gcv -nostdlib -nostartfiles -Trv32v_simple_selfcheck_0000.ld -o rv32v_simple_selfcheck_0000.o rv32v_simple_selfcheck_0000.s

Test execution via Spike:
spike --isa=RV64IMAFDCV -l -p1 v32v_simple_selfcheck_0000.o

aswaterman, 2019-10-11:
@chihminchao

chihminchao, 2019-10-12:
Hi  Chert-Nik,
The segfault is expected If there is out of range index access for vector register.  For current implementation and performance consideration, there is no strict checking about whether the target program access vector register or element correctly.
I take quick look at the testing code. 
LMUL is set to 4 by 
`vsetvli t0, a0, e32, m4  #103`
than the max register index could be used is  28 and the following code sequences use out of range vector register just like the line where you get segfault

Thanks

Chert-Nik, 2019-10-12:
Just looked at [RISC-V "V" Vector Extension](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc) specification.

Section _3.2.2 Vector Register Grouping (vlmul)_ says that:

> Similarly, when vlmul=10, vector instructions operate on four vector registers at a time, and instructions specifying vector operands using vector register numbers that are not multiples of four will raise an illegal instruction exception.

This register alligment check will fix issue with current segfault situation.
Because segmentation fault is not an option for any program, i can think of two solutions:

First (the easy one, greater impact on perfomance):
1. Add register alligment check inside `vectorUnit_t::elt()` function

Second (more work to do, less impact on perfomance):
1. Add new macro in **decode.h** (lets call it **VI_RD** for now), that do the following:
a) retrieves the destination register from insn_t by `insn_t::rd()`
b) performs checks based on selected grouping mode. If alligment requirement is not met, than `throw trap_illegal_instruction(0)`.
2. Replace **VI_LD** macro line at **decode.h:1433**
from `const reg_t vd = insn.rd(); \`
to `const reg_t vd = VI_RD; \`
3. Do the same kind of thing at all other places in **decode.h** where `P.VU.elt()` function is called

That way conformance with the specification can be achieved.

chihminchao, 2019-10-14:
What you say is right  and the second solution is preferable. The spec requires lots of unexpected usage and triggers illegal exceptions  However, the first release mostly focuses the expected behavior and misses certain of this kind checking.  We are working to complete them.

chihminchao, 2019-11-18:
@Chert-Nik 
Register usage checker has been merged into master branch.  That should resolve your issue.

closed 2019-11-18 by Chert-Nik