riscv-software-src/riscv-isa-sim #347: Unable to build Dummy-RoCC extension

fkan-ec, 2019-10-12:
$ spike --extension=dummy_rocc                                                                                                                                                             
couldn't find extension 'dummy_rocc' (or library 'libdummy_rocc.so')

fkan-ec, 2019-10-13:
I don't actually see any dummy_rocc.so built. The dummy_rocc.cc is also not compiled. Can anyone help me with this? Much Thanks.

a0u, 2019-10-13:
Subprojects are statically linked now, so shared libraries are not longer built by default.

Here is a patch for dummy_rocc (apply with `git am`):
https://gist.github.com/a0u/e7fd476ecb6981a8416108ec3c18e389

This isn't an ideal solution as `libriscv.a` has to be built with `-fPIC`, which potentially has an adverse performance impact.

fkan-ec, 2019-10-15:
After rebuilding the code, I did a make libdummy_rocc.so and was able to generate a libdummy_rocc.so. However, running was still a problem. It seem to dlopen the shared library but unable to register the extension.

$ spike --extension=dummy_rocc pk                                                                                        
couldn't find extension 'dummy_rocc' in shared library 'libdummy_rocc.so'

HolyLow, 2020-06-10:
Same problem, still not solved yet @ @a0u @fkan-ec

HolyLow, 2020-06-11:
@chihminchao I tried the #485 , but still
`build git:(master) ✗ spike --extension=dummy_rocc $RISCV/riscv64-unknown-elf/bin/pk a.out
bbl loader
z  0000000000000000 ra 0000000000010102 sp 000000007f7e9b00 gp 0000000000015ab8
tp 0000000000000000 t0 0000000000010454 t1 000000000000000f t2 0000000000000000
s0 000000007f7e9b50 s1 0000000000000000 a0 0000000000000001 a1 000000000000dead
a2 0000000000000001 a3 000000000000beef a4 0000000000000bad a5 0000000000000001
a6 000000000000001f a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 0000000000000000 t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
pc 0000000000010188 va 0000000000c5f50b insn       00c5f50b sr 8000000200046020
An illegal instruction was executed!`
I also tried to add some printf to dummy_rocc_t constructor and functions, but nothing is printed(I am not sure if the stderr is redirected and discarded, or the dummy_rocc extension is never constructed which means the extension is not activated).
Another possibility lies on the pk I use (which is compiled without any modification). However, I don't know what I should modify or anything else.

chihminchao, 2020-06-12:
It works and the opcode "00c5f50b" is illegal instruction because it hit https://github.com/riscv/riscv-isa-sim/blob/983abeaa9fb9c9c5cc8b318fdad11305a88021f3/customext/dummy_rocc.cc#L15.
But the disassemble doesn't handle extension initialization from "--extension" path well.  I will send another PR to fix it.

HolyLow, 2020-06-15:
@chihminchao I changed the dummy_rocc behavior in my local repository so that the instruction 0x00c5f50b won't fall into an illegal instruction. But it still doesn't work, and I think the "--extension" initialization is not working (because my dummy_rocc_t is even not constructed).
Looking forward to your PR, thanks a lot!!

HolyLow, 2020-06-15:
@chihminchao I solved my problem! It was the issue of my pk, I should set csv to enable xcustom instructions in my used pk.
Thanks a lot for spending your time on the issue!
Besides, the provided dummy_rocc_test.cc will not get compiled successfully by riscv64-unknown-elf-gcc (nor the gnu version). I am planning to send a PR for a ready-to-use dummy_rocc version which adopts the solution in the toolchain issue https://github.com/riscv/riscv-gnu-toolchain/issues/190.

dyazzvlue, 2021-12-15:
@chihminchao  I am a rookie of spike. I tried to add a new customext like cflush.cc in #485 . Add "my_ext.cc" to **customext_srcs** in customext.mk.in and rebuild the simulator. After that , i tried below:
$./build/spike --extension=my_ext pk test.out
couldn't find extension 'cosim_rocc' in shared library 'libcustomext.so'
Do you have any idea about this?