riscv-software-src/riscv-isa-sim #348: Running "make install" a Second Time Fails Because they are read-only

joelsherrill, 2019-10-17:
This is as simple bug in the Makefile.in. Header files should be installed as 0644 not 0444.  

When you run make install a second time, you do not have permission to rewrtie the .h files.

The line to change is: https://github.com/riscv/riscv-isa-sim/blob/master/Makefile.in#L114

joelsherrill, 2019-10-17:
From 6c0a2759583a4f433f7b56ac1f9a4ea564da19f6 Mon Sep 17 00:00:00 2001          
From: Joel Sherrill <joel@rtems.org>
Date: Thu, 17 Oct 2019 08:18:32 -0500
Subject: [PATCH] Correct Permission on Installed Headers

Fixes Issue #348
---
 Makefile.in | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/Makefile.in b/Makefile.in
index 605cb52..49e1857 100644
--- a/Makefile.in
+++ b/Makefile.in
@@ -111,7 +111,7 @@ RUNFLAGS      := @RUNFLAGS@
 
 MKINSTALLDIRS := $(scripts_dir)/mk-install-dirs.sh
 INSTALL       := @INSTALL@
-INSTALL_HDR   := $(INSTALL) -m 444
+INSTALL_HDR   := $(INSTALL) -m 644
 INSTALL_LIB   := $(INSTALL) -m 644
 INSTALL_EXE   := $(INSTALL) -m 555
 STOW          := @stow@
-- 
1.8.3.1

aswaterman, 2019-10-18:
Fixed in fb27391bf65ed867b62ef40b726a21ac839fa37a

closed 2019-10-18 by aswaterman

joelsherrill, 2019-10-24:
Thanks for fixing this and also catching the executables not being 755. I had updated the RTEMS local patch but was traveling and didn't update it here.

Thanks for being so responsive. We are using Spike for testing RTEMS.

aswaterman, 2019-10-24:
Ah, cool, good luck with that.