riscv-software-src/riscv-isa-sim #349: Misaligned instruction access

linjlpp, 2019-10-18:
Hi,

I am compiling the _hello world_ with riscv64-unknown-linux-gnu-g++ (instead of riscv64-unknown-elf-g++). Then I got the error message of **"Misaligned instruction access"**. Is there any solution to resolve this?

>  z  0000000000000000 ra 00000000000103d4 sp 000000007f7e7b50 gp 0000000000012830
tp 0000000000000000 t0 0000000000000000 t1 0000000000000010 t2 0000000000012380
s0 0000000000000000 s1 0000000000000000 a0 00000000000104b4 a1 0000000000000001
a2 000000007f7e7b58 a3 000000000001050c a4 000000000001059c a5 0000000000000000
a6 000000007f7e7b50 a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 ffffffffffffffff t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
pc 000000000001039c va fffffffffffffffe insn       ffffffff sr 8000000200046020
Misaligned instruction access!

aswaterman, 2019-10-18:
Did you forget to use the `-static` flag?  The Linux compiler dynamically links by default.

linjlpp, 2019-10-18:
> Did you forget to use the `-static` flag? The Linux compiler dynamically links by default.

Thanks @aswaterman , this fixes the problem.

aswaterman, 2019-10-18:
👍

closed 2019-10-18 by aswaterman