riscv-software-src/riscv-isa-sim #350: Risc-v vector extension

allenderalencar, 2019-10-20:
I`m trying to run some examples from various sources and always get:
"core   0: exception trap_instruction_access_fault, epc 0x0000000000000000
core   0:           tval 0x0000000000000000"

*I tried from: https://github.com/riscv/riscv-gnu-toolchain/issues/460
the example:

- .global main
- main:
- vsetvli t0, x0, e32
- vadd.vv v0, v0, v0 # changed from vadd to vadd.vv
- ret

-run: /path/to/in/bin/riscv64-unknown-elf-as -march=rv64gcv -mabi=lp64f wya.s -o wya.o 
       /opt/isaa/bin/spike --isa=RV64IMAFDCV -l -p1 wya.o

*Tried from the risc-v vector documentation

- loop:
-  vsetvli a3, a0, e32,m8 # Use only 32-bit elements
-  vlh.v v8, (a1) # Sign-extend 16b load values to 32b elements
-  sll t1, a3, 1 # Multiply length by two bytes/element
-  add a1, a1, t1 # Bump pointer
-  vmul.vx v8, v8, x10 # 32b multiply result
-  vsrl.vi v8, v8, 3 # Shift elements
-  vsw.v v8, (a2) # Store vector of 32b results
-  sll t1, a3, 2 # Multiply length by four bytes/element
-  add a2, a2, t1 # Bump pointer
-  sub a0, a0, a3 # Decrement count
-  bnez a0, loop # Any more?

-run: /path/to/in/bin/riscv64-unknown-elf-as -march=rv64gcv -mabi=lp64f cv.s -o cv.o                                          
       /opt/isaa/bin/spike --isa=RV64IMAFDCV -l -p1 cv.o 

*And can't compile: https://github.com/ispras/riscv-avs/tree/master/testsuite/synthetics/rv32v
For some errors like:
rv32v_simple_selfcheck_0000.s:109: Error: illegal operands `vlw.v v0,t1'
rv32v_simple_selfcheck_0000.s:111: Error: illegal operands `vlw.v v1,t1'

-run: /path/to/in/bin/riscv64-unknown-elf-gcc -march=rv64gcv -nostdlib -nostartfiles rv32v_simple_selfcheck_0000.ld -o test.o rv32v_simple_selfcheck_0000.s 

As suggested, I tried too hello.c:

- #include <stdio.h>
- int main()
- {
-    printf("Hello, World!");
-    return 0;
- }

-run: /path/to/in/bin/riscv64-unknown-elf-gcc -o hello hello.c
        /opt/isaa/bin/spike pk hello

and get:
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not open pk (did you misspell it? If VCS, did you forget +permissive/+permissive-off?)

If you have some vector example that works and can be compiled or can help with some hint, I would be very grateful, thank you for your attention.

aswaterman, 2019-10-20:
> *I tried from: riscv/riscv-gnu-toolchain#460

You’re trying to execute an object file, rather than a linked program. See
the README for an example of how to build a linked “hello world” program
that can run on Spike.

allenderalencar, 2019-10-20:
Thx for your reply, as I said, I tried the hello world too:
"As suggested, I tried too hello.c:

#include <stdio.h>
int main()
{
printf("Hello, World!");
return 0;
}
-run: /path/to/in/bin/riscv64-unknown-elf-gcc -o hello hello.c
/opt/isaa/bin/spike pk hello

and get:
terminate called after throwing an instance of 'std::runtime_error'
what(): could not open pk (did you misspell it? If VCS, did you forget +permissive/+permissive-off?)"
Do you know what is causing that problem? 
The spike has some documentation that describe message error?
thank you for your attention.

aswaterman, 2019-10-20:
It looks like you are missing pk (
https://github.com/riscv/riscv-pk)

jinchenglee, 2019-10-21:
@allenderalencar , not sure it is helpful. I found sometimes you have to specify the full path to the right version (32/64) of pk while you call spike.

luismarques, 2019-10-25:
Indeed. I use a custom installed that installs a `pk32` and a `pk64` binary and I use those names in the spike command-line to avoid ambiguities. But ideally spike / the standard install would be improved to better handle that issue, so that workarounds wouldn't be needed.

gsauthof, 2020-02-02:
Looks like this issue is resolved and can be closed.

FWIW, I've recently used Spike with some "V" 0.8 code and pk and it worked fine.

closed 2020-02-03 by aswaterman