riscv-software-src/riscv-isa-sim #357: Number of cycles in RISC-V

wyahiaoui, 2019-11-17:
In the goal to measure the performance of some Applications after its optimization. I tried to calculate the execution time between the relevant part of code, but apparently it results the performance of the simulator and not of the optimized program. I had the idea to count the number of cycles instead of.
I am using 

unsigned long read_cycles(void)
{
    unsigned long cycles;
    asm volatile ("rdcycle %0" : "=r" (cycles));
    return cycles;
}

to get the number of cycles, it compiles fine with GCC but it crashes directly when I execute it by spike. What should be wrong here? and I am asking whether there is an alternative to get the speedup of the optimized code. 

Thanks in advance

aswaterman, 2019-11-17:
Seems to work for me:

```
$ cat test.c
#include <stdio.h>

unsigned long read_cycles(void)
{
  unsigned long cycles;
  asm volatile ("rdcycle %0" : "=r" (cycles));
  return cycles;
}

int main()
{
  printf("%ld\n", read_cycles());
  printf("%ld\n", read_cycles());
  return 0;
}
$ riscv64-unknown-elf-gcc -O2 test.c
$ spike pk a.out
bbl loader
250591
298699
```

aswaterman, 2019-11-17:
(If you are running bare-metal in machine mode, you might try `csrr %0, mcycle` instead.)

wyahiaoui, 2019-11-17:
Yes, That's exactly the point, it works when SPIKE is executed without --isa=rv64gcv and actually It's needed for the V-extension. 

Do you have an idea?

aswaterman, 2019-11-18:
That seems to work, too:

```
$ spike --isa=rv64gcv pk a.out
bbl loader
250591
298699
```

vincentabraham, 2021-11-11:
> Seems to work for me:
>
> ```
> $ cat test.c
> #include <stdio.h>
>
> unsigned long read_cycles(void)
> {
>   unsigned long cycles;
>   asm volatile ("rdcycle %0" : "=r" (cycles));
>   return cycles;
> }
>
> int main() [...]

How could you find out the actual delay from machine cycles later on ?