riscv-software-src/riscv-isa-sim #358: Vector load/store instructions don't work

xicongye, 2019-11-19:
Hello, I try to run some programs on spike, however, spike's behavior didn't match my expectations. 
My steps are as follows: 
1. build spike
```
$ git clone https://github.com/riscv/riscv-isa-sim.git
$ cd riscv-isa-sim
$ mkdir build && cd build
$ ../configure  --with-isa=rv32imafcv --with-varch=v32:e32:s32 --enable-histogram
$ make -j$(nproc) && make install
```
2. build pk
```
$ git clone https://github.com/riscv/riscv-pk.git
$ cd riscv-pk
$ mkdir build && cd build
$ ../configure  --host=riscv64-unknown-elf --with-arch=rv32imac
$ make -j$(nproc) && make install
```
3. write a simple program to test spike:
```
$ cat main.c
#include <stdio.h>
uint32_t x[8] = {1, -1, 2, 3, 4, 5, 6, 7};
uint32_t y[8] = {0};

int main(void)
{
  asm(
       "vsetvli    a1, x0, e32, m8 \n"
       "vlw.v      v8, (%0)        \n"
       "vsw.v      v8, (%1)       \n"
       :   
       : "r"  (x), "r" (y) 
   );  

   for(int i = 0; i < 8; i++)
      printf("y[%d] = 0x%x\n", i, y[i]);
  return 0;
}
```

4. compile & run 
```
$ riscv64-unknown-elf-gcc -O0 -march=rv32imafcv -mabi=ilp32f main.c -o main.elf
$ spike pk main.elf
bbl loader
y[0] = 0x0
y[1] = 0x0
y[2] = 0x0
y[3] = 0x0
y[4] = 0x0
y[5] = 0x0
y[6] = 0x0
y[7] = 0x0
```

Actually, what I expect is like this:
```
y[0] = 0x1
y[1] = 0xffffffff
y[2] = 0x2
y[3] = 0x3
y[4] = 0x4
y[5] = 0x5
y[6] = 0x6
y[7] = 0x7
```
However, when I use other branch of spike, like 'rvv-vnclip', it works fine.

aswaterman, 2019-11-19:
>    1. build pk
>    1. write a simple program to test spike:
>    1. compile & run

The proposed definition of vsetvli has changed a couple times recently,
such that in some versions, using rs1=x0 doesn't ask for MAXVL, but instead
preserves whatever the current vector length is.  Most likely you'll get
what you're looking for if you explicitly ask for a vector length of 8
instead of x0.

xicongye, 2019-11-19:
Thanks, it works now.

closed 2019-11-19 by xicongye