riscv-software-src/riscv-isa-sim #359: riscv-gdb/gdb/source-cache.c:216: undefined reference to `srchilite::SourceHighlight::SourceHighlight(...

blastwave, 2019-11-20:
Essentially trying to build on a FreeBSD amd64 system while using the native gcc 9.2.0 pkgs I see : 

  CXXLD  gdb
/usr/local/bin/ld: source-cache.o: in function `source_cache::get_source_lines(symtab*, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*)':
/opt/rv64/tool_build/riscv-gnu-toolchain/riscv-gdb/gdb/source-cache.c:216: undefined reference to `srchilite::SourceHighlight::SourceHighlight(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/local/bin/ld: /opt/rv64/tool_build/riscv-gnu-toolchain/riscv-gdb/gdb/source-cache.c:220: undefined reference to `srchilite::SourceHighlight::highlight(std::istream&, std::ostream&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
gmake[3]: *** [Makefile:1889: gdb] Error 1
gmake[3]: Leaving directory '/opt/rv64/tool_build/riscv-gnu-toolchain/build-gdb-newlib/gdb'
gmake[2]: *** [Makefile:9126: all-gdb] Error 2
gmake[2]: Leaving directory '/opt/rv64/tool_build/riscv-gnu-toolchain/build-gdb-newlib'
gmake[1]: *** [Makefile:849: all] Error 2
gmake[1]: Leaving directory '/opt/rv64/tool_build/riscv-gnu-toolchain/build-gdb-newlib'
gmake: *** [Makefile:410: stamps/build-gdb-newlib] Error 2
vesta# 

Perhaps I need to specify the C++ compiler in an env var CXX ?

MengshiLi, 2020-02-02:
Hit the same issue, did you find a solution to it?
Thanks

melak, 2025-03-20:
Hit the same issue (although with `devel/gcc-arm-embedded`, but I don't suppose they are all that different from this perspective). In my case the solution was to rebuild `devel/gdb` with `SOURCE_HIGHLIGHT=off`, uninstall `textproc/source-highlight`, and then `devel/gcc-arm-embedded` suddenly started to behave.