riscv-software-src/riscv-isa-sim #361: Bogus "bad --isa option"

scottj97, 2019-11-25:
In change 0162a6e72421b5cbec1905b4cae7bfab98afe83f, `parse_isa_string()` changed so it sets only `max_isa` and not `state.misa`. But at the end of that function, it calls `supports_extension()` which checks `state.misa`, which is uninitialized, leading to random bogus failures such as:

```
error: bad --isa option RV64IMAC
```

aswaterman, 2019-11-25:
Thanks for the bug report.  The problem is that `state.misa` just isn't initialized at that point, so the behavior is undefined.  As luck would have it, it always happens to get zero-initialized in my use cases, so I can't see the bug without forcibly initializing the memory to some other value.

aswaterman, 2019-11-25:
I marked this as closed in the referenced commit, but please reopen if it doesn't fix your use case.

closed 2019-11-25 by aswaterman

scottj97, 2019-11-25:
Works great, thank you!

commit b36b0d66 (2019-11-25): Initialize state.misa prior to calls to supports_extension

Partially reverts 0162a6e72421b5cbec1905b4cae7bfab98afe83f

Closes #361