riscv-software-src/riscv-isa-sim #363: valgrind: clint->mtime uninitialized

scottj97, 2019-11-27:
Running same command as #362 from newly-updated master 1f34121c901164b6e11d4190833026224a6ff1cb, valgrind reports:

```
==29170== Conditional jump or move depends on uninitialised value(s)
==29170==    at 0x48C283: clint_t::increment(unsigned long) (clint.cc:69)
==29170==    by 0x44E86B: sim_t::step(unsigned long) (sim.cc:122)
==29170==    by 0x43AF98: sim_t::main() (sim.cc:94)
==29170==    by 0x5B3B00F: ??? (in /usr/lib64/libc-2.17.so)
==29170==    by 0x5FF7C2F: ???
==29170==  Uninitialised value was created by a heap allocation
==29170==    at 0x4C2A1E3: operator new(unsigned long) (vg_replace_malloc.c:334)
==29170==    by 0x44F475: sim_t::sim_t(char const*, char const*, char const*, unsigned long, bool, unsigned long, std::vector<std::pair<unsigned long, mem_t*>, std::allocator<std::pair<unsigned long, mem_t*> > >, std::vector<std::pair<unsigned long, abstract_device_t*>, std::allocator<std::pair<unsigned long, abstract_device_t*> > >, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<int, std::allocator<int> >, debug_module_config_t const&) (sim.cc:68)
==29170==    by 0x43A863: main (spike.cc:252)
```

Which seems to be because `mtime` is not initialized.

aswaterman, 2019-11-28:
Thanks, keep 'em coming

closed 2019-11-28 by aswaterman

scottj97, 2019-11-28:
Thanks. Valgrind is all clean now! I also confirmed that it correctly detects #361.

aswaterman, 2019-11-28:
Cool, thanks for your help squashing these.

commit 77661f72 (2019-11-28): Initialize mtime

Closes #363