riscv-software-src/riscv-isa-sim #364: How do tohost and fromhost work?

terabyte128, 2019-12-01:
I defined `tohost` and `fromhost`as memory locations in the linker script for a bare-metal program that I'm trying to run within spike. But I can't find anywhere how these symbols are actually used. I assumed that I could just write bytes to the location as defined by `tohost` and they would show up, but that doesn't seem to be the case.

How can I use these symbols to actually exchange data between spike and the host machine?

scottj97, 2019-12-02:
Months ago I unsuccessfully tried to find the same info.

What I do understand:

Writing a 32-bit value to tohost with lsb=0 triggers a "syscall", though this mechanism doesn't seem to be used in the isa tests. 

Writing a 32-bit value to tohost with lsb=1 terminates the simulation. The upper 31 bits are the exit code: 0=PASS, anything else=FAIL.

Spike detects this write to tohost address and uses it to terminate the program with either a pass or fail condition. Formerly this was in a separate riscv-fesvr ("frontend server") package.

I haven't tried to figure out fromhost, nor do any of the isa tests use this, that I can tell.

ccelio, 2019-12-02:
`fromhost`, as I remember, is needed for the design-under-test to poll the host system (running the "frontend server" or fesvr) when proxying a syscall onto the fesvr. I think it basically just looks for non-zero values, but it's just whatever protocol the fesvr software feels like using.  

I guess you could use `fromhost` to do more powerful stuff like providing inputs/loading binaries into the DUT, but generally it's a good idea to have some other mechanism for the host system to talk to the

closed 2019-12-02 by ccelio

scottj97, 2019-12-02:
Not sure why this was closed unceremoniously. Consider this a documentation request. Since riscv-fesvr was folded in to Spike, this seems like the right repo for it.

When you say "it's just whatever protocol the fesvr software feels like using", isn't Spike that fesvr software now?

scottj97, 2019-12-17:
I noticed in [riscv-test-env/v/vm.c function `do_tohost()`](https://github.com/riscv/riscv-test-env/blob/99eb76ac3c19eb3ebc6c2f91c4bfb85baf187cd0/v/vm.c#L25-L27) there is code that reads from `tohost` and writes to `fromhost`, which seems backwards to me.

Also in that same file [function `cputchar()`](https://github.com/riscv/riscv-test-env/blob/99eb76ac3c19eb3ebc6c2f91c4bfb85baf187cd0/v/vm.c#L43) uses some magic value `0x0101000000000000`, which doesn't seem to be explained anywhere. It also seems strange to me that it writes the char without shifting it left by 1 bit, since lsb=1 indicates test exit (as I'd explained above). Wouldn't this trigger simulation exit with an error code half of the time (any odd-valued character)?

kathlenehurt-sifive, 2020-03-11:
I'm also trying to understand this functionality.
- Is this the only method of ending a test that Spike supports?
- Where can I decode the "magic values" written to tohost?
- Is this functionality specific to Spike? Or RISC-V?

chaoyangnz, 2020-04-02:
Really lack of documentation. Just curious how Spike developers don’t confuse themselves. I searched everywhere on the internet, just found the read me, nothing else. The same thing applies to other knowledge like plugins. If the developers of Spike assume the users know Spike like them, please ignore my comments, otherwise it is a shame to do fundamental work without documentation.

aswaterman, 2020-04-02:
Fine, I'll take the bait.  This was an experiment that got out of the lab.  The HTIF is a tether between a simulation host and target, not something that's supposed to resemble a real hardware device.  It's not a RISC-V standard; it's a UC Berkeley standard.  You can still use Spike without using HTIF; you just need to link in your own I/O devices instead.

Here is a brief sketch of the tohost format, excluding some stuff that I don't think is immediately relevant:

- Bits 63:56 indicate the "device".
- Bits 55:48 indicate the "command".
- Device 0 is the syscall device, which is used to emulate Unixy syscalls.  It only implements command 0, which has two subfunctions (for legacy reasons, sorry for the bad design):
  - If bit 0 is clear, then bits 47:0 represent a pointer to a struct describing the syscall.  The format of the syscall struct is pretty clear if you read the pk or fesvr code.
  - If bit 1 is set, then bits 47:1 represent an exit code, with a zero value indicating success and other values indicating failure.
- Device 1 is the blocking character device.
  - Command 0 reads a character
  - Command 1 writes a character from the 8 LSBs of tohost

To answer @scottj97's question, the reason the LSB being set in his example didn't exit the program is that the command was targeted towards the blocking character device, not towards the syscall device.

scottj97, 2020-06-11:
You can see tohost/fromhost in use in OpenSBI: [htif.c](https://github.com/riscv/opensbi/blob/master/lib/utils/sys/htif.c)

timsifive, 2021-03-18:
> You might notice the 64-bit orientation. For RV32, the tohost register is zero-extended, so only device=0 and command=0 (i.e. HTIF syscalls/exit codes) are supported.

That doesn't appear to be the case. https://github.com/riscv/riscv-isa-sim/blob/a31ac28aff73dafa41c9968452df95647cf0ae8c/fesvr/htif.cc#L219 reads a 64-bit word from memory regardless of XLEN. The problem is that on RV32 you can't do an atomic 64-bit write, so htif might only see half the word you wrote. In practice, however, it appears to work fine. I assume this is because spike executes batches of 5000 instructions at a time, so your program would have to be quite unlucky for the 2 consecutive stores to be the boundary of the 5000 instructions.

@aswaterman, would you accept a spike change that changes the htif logic to only read tohost if any of the low 32 bits set? In that case RV32 can write the high bits first, and not have to worry about this race.

aswaterman, 2021-03-18:
That only works if we forbid messages with the lower 32 bits all zero, which isn't backwards compatible.  Given the lack of HTIF device emulation support for RV32 doesn't seem like a huge problem in practice, I'm not inclined to go down that route.  I'll see if I can think of a related hack that can be implemented compatibly.

timsifive, 2021-03-19:
Maybe not compatible, but I don't know of any reason you'd want to. There are 2 devices:
1. syscall, in which case the payload either has the LSB set to exit, or it's a pointer. I imagine that pointer will always be non-zero and 4-byte aligned.
2. Character device, in which case the payload is the character to write. Writing NUL to a terminal seems rarely useful.

I am using this mechanism in the FreeRTOS demo running on spike, so it can do more or less the same thing that the FreeRTOS demo does on Qemu. Maybe proxy syscalls are a better mechanism to use?

aswaterman, 2021-03-19:
(Well, writing NUL does happen, even if it's not useful.)

I suggest using the proxy syscalls; they are straightforward to set up and can probably provide the functionality you need.

ewtoombs, 2026-10-13:
Here's a complete example:
```
.data
hello:
.string "hello\n"
.align 2
write_struct:
    .8byte 64, 1, hello, 6, 0, 0, 0, 0

.text
.global _start
_start:
    la t1, write_struct
    sw t1, tohost, t2
/*
    li t1, 1
    sw t1, tohost, t2
*/
```
It is supposed to write "hello" to stdout, then exit.

Link with this linker script:
```
MEMORY
{
    ram (rwx) : ORIGIN = 0x80000000, LENGTH = 16M
}

SECTIONS
{
    .text : {
        *(.text._start)
        *(.text*)
    }
    .data : {
        *(.data*) 
    }
    . = ALIGN(32);
    tohost = .;
    . += 32;
    fromhost = .;
}
```

```
> riscv32-elf-as add.s -o add.o
> riscv32-elf-ld --script spike.ld -o add add.o
> spike --isa=RV32I --priv=M add
```

It doesn't work though, unless the exit command is left out. 

I'm opening this as a separate bug.

aswaterman, 2026-10-13:
The effect of writing `tohost` is not instantaneous.  The host only polls that memory occasion periodically.  The protocol is that you need to spin until the host writes a nonzero value to `fromhost`; see e.g. https://github.com/riscv-software-src/riscv-tests/blob/7b68f7b91f827f8d94cc1c57ff123752c5c8f704/benchmarks/common/syscalls.c#L29-L32

ewtoombs, 2026-10-14:
Oh. In that case, do I also have to write zero to `fromhost` before writing to `tohost`?

ewtoombs, 2026-10-14:
And what is this `__sync_synchronize` thing?

ewtoombs, 2026-10-14:
OK, *here's* a complete example:
```
.data
hello:
.string "hello\n"
.align 2
write_struct:
    .8byte 64, 1, hello, 6, 0, 0, 0, 0

.text
.global _start
_start:
    la t1, write_struct
    sw t1, tohost, t2
wait:
    lw t1, fromhost
    beqz t1, wait

    slli t1, t1, 1
    addi t1, t1, 1
    sw t1, tohost, t2
```
It writes "hello" to stdout, then exits with the return value of the write syscall.

Link with this linker script:
```
MEMORY
{
    ram (rwx) : ORIGIN = 0x80000000, LENGTH = 16M
}

SECTIONS
{
    .text : {
        *(.text._start)
        *(.text*)
    }
    .data : {
        *(.data*) 
    }
    . = ALIGN(8);
    tohost = .;
    . += 8;
    fromhost = .;
}
```

```
> riscv32-elf-as add.s -o add.o
> riscv32-elf-ld --script spike.ld -o add add.o
> spike --isa=RV32I --priv=M add
```

aswaterman, 2026-10-14:
> Oh. In that case, do I also have to write zero to fromhost before writing to tohost?

I think the ABI assumes `fromhost` is initially zeroed, IOW, it is normally zeroed after the transaction, not before.  But either way works, I guess.

> And what is this __sync_synchronize thing?

This is the GCC builtin for a memory fence.  That code predates C11; these days you'd write `atomic_thread_fence(memory_order_seq_cst)`.

ewtoombs, 2026-11-04:
I tried using the character device.
I kinda hate it, because a delay is required, but spike doesn't set fromhost to a non-zero value upon completion, so I have to just guess at how long it takes to complete.
This works:
```
u32 __attribute__((align(8))) tohost[2];
void char_write(char c)
{
    tohost[1] = 1 << 56-32 | 1 << 48-32;
    tohost[0] = (u32)(c);
    for (u32 i = 0; i < 5000; i++) ; 
}
```

Without that for loop, it fails.
Actually, the entire character device, I think, is kinda dumb, and should be deprecated, then removed.

Also, couldn't spike tell when `tohost[0]` is written to, even if it is just 0 that is written to it?
Couldn't spike differentiate a write to `tohost[0]` from a write to `tohost[1]`?

ewtoombs, 2026-11-04:
I started documenting this in https://github.com/riscv-software-src/riscv-isa-sim/pull/2528 .