riscv-software-src/riscv-isa-sim #368: pmp check don't match riscv privileged spec.

weiqi7777, 2019-12-06:
below is the check pmp code.

```c++
reg_t mmu_t::pmp_ok(reg_t addr, reg_t len, access_type type, reg_t mode)
{
  if (!proc)
    return true;

  reg_t base = 0;
  for (size_t i = 0; i < proc->state.n_pmp; i++) {
    reg_t tor = proc->state.pmpaddr[i] << PMP_SHIFT;
    uint8_t cfg = proc->state.pmpcfg[i];

    if (cfg & PMP_A) {
      bool is_tor = (cfg & PMP_A) == PMP_TOR;
      bool is_na4 = (cfg & PMP_A) == PMP_NA4;

      reg_t mask = (proc->state.pmpaddr[i] << 1) | (!is_na4);
      mask = ~(mask & ~(mask + 1)) << PMP_SHIFT;

      // Check each 4-byte sector of the access
      bool any_match = false;
      bool all_match = true;
      for (reg_t offset = 0; offset < len; offset += 1 << PMP_SHIFT) {
        reg_t cur_addr = addr + offset;
        bool napot_match = ((cur_addr ^ tor) & mask) == 0;
        bool tor_match = base <= cur_addr && cur_addr < tor;
        bool match = is_tor ? tor_match : napot_match;
        any_match |= match;
        all_match &= match;
      }

      if (any_match) {
        // If the PMP matches only a strict subset of the access, fail it
        if (!all_match)
          return false;

        return
          (mode == PRV_M && !(cfg & PMP_L)) ||
          (type == LOAD && (cfg & PMP_R)) ||
          (type == STORE && (cfg & PMP_W)) ||
          (type == FETCH && (cfg & PMP_X));
      }
    }

    base = tor;
  }

  return mode == PRV_M;
}
```

for rv64 m-mode, it execute the ld ins, load 8 byte data from the address 0x90000000.

for  pmp entry10 
* config  is 0x15
  * math mode is NA4
* pmpaddr[10] is 0x24000000
  * protect the base address is 0x90000000

  
for the entry10 pmp check, any_match is true, but all_match is false. that cause pmp check failed.

so in m-mode, execute the ld ins, it will get the  trap_load_access_fault.

from the riscv-privileged spec,  for pmp check in m-mode,  if the lock bit is clear, the pmp entry is  succeed. so in this , m-mode executing the ld ins shouldn't get the trap.

can you have a look?
thanks

aswaterman, 2019-12-06:
The spec says, "The matching PMP entry must match all bytes of an access, or the access fails, irrespective of the L, R, W, and X bits."

An 8-byte access that overlaps a 4-byte region should fail.

weiqi7777, 2019-12-08:
thanks for you replay
i will close this issue.

closed 2019-12-08 by weiqi7777