riscv-software-src/riscv-isa-sim #369: Question：How to change the softfloat to 3e version？

foofoofoolishlearner, 2019-12-11:


aswaterman, 2019-12-11:
Any particular reason you want to do so?

We had to modify softfloat a bit to make it work in spike, so it is probably not straightforward.

foofoofoolishlearner, 2019-12-11:
There is a problem that confuses me.  If the opcode is 0x191676d3，the instruction is fdiv.s fa3, fa2, fa7. Operanda fa2 :ffffffff4f00012f;  operandb fa7 :ffffffff0000002f. This arithmetic result should be out of range, and the value should be ffffffff7f800000. But the result of compiled spike is ffffffff7f7fffff. So I think it may be a version problem or there are parameters when compiling I did not add. I don't know much about spike, and thank you for your prompt reply.

aswaterman, 2019-12-11:
Thanks for the explanation. I’ll look into this tomorrow.

aswaterman, 2019-12-12:
I am seeing correct behavior for this case:

```: freg 0 fa5
0xffffffffffffffffffffffff4f00012f
: freg 0 fa4
0xffffffffffffffffffffffff0000002f
: 
core   0: 0x00000000000100ee (0x18e7f7d3) fdiv.s  fa5, fa5, fa4
: freg 0 fa5
0xffffffffffffffffffffffff7f800000
: 
```

foofoofoolishlearner, 2019-12-12:
My result：

: freg 0 fa3       
0xffffffffffffffff41ec95e284000000
: freg 0 fa2
0xffffffffffffffffffffffff4f00012f
: freg 0 fa7
0xffffffffffffffffffffffff0000002f
: 
core   0: 0x000000008000939e (0x191676d3) fdiv.s  fa3, fa2, fa7
------------------ Mode:2 ------------------
------------------ RES:7f7fffff ------------------
--------------- EXCEPT: 5 ---------------
--------------- FFLAGS: 15 ---------------
: freg 0 fa3
0xffffffffffffffffffffffff7f7fffff

And how I install the spike：

 1003  ../configure --prefix=/home/cwu/gitlist/riscv-isa-sim/cwutool1
 1004  make
 1005  make install

Maybe there is something wrong with my installation.

aswaterman, 2019-12-12:
Interesting. Can you describe your host machine? I am using x86-64 Linux.

foofoofoolishlearner, 2019-12-12:
Mine is Linux 3.10.0-1062.4.3.el7.x86_64 x86_64.

aswaterman, 2019-12-12:
Sounds like you need to do some more investigation. Please do not simply
try to upgrade softfloat. My experiment indicates that will not help.

foofoofoolishlearner, 2019-12-13:
Can you send me the elf file you run？
I want have a try on my machine.

aswaterman, 2019-12-13:
[a.out.zip](https://github.com/riscv/riscv-isa-sim/files/3962301/a.out.zip)

aswaterman, 2019-12-13:
Sorry for using .zip; that's what GitHub told me I had to do...

foofoofoolishlearner, 2019-12-16:
Thank you so much for your patience. And I had found my problem: an instruction changed the value of the frm register before this fdiv instruction was executed, so the rounding mode became softfloat_round_min and the result became 7f800000-1=7f7fffff. 
Is it correct to think so? I just thought that the result should be infinite, without considering the effect of the round_min. Sorry for asking naive questions.

aswaterman, 2019-12-16:
That computation is too far out of range for rounding to matter.  I think you've got something else going on.  Reopen if you think there's a bug in spike.

closed 2019-12-16 by aswaterman

smtdta, 2020-02-23:
hi, can you tell me are you getting result ff7fffff if rounding mode is changed to round_max instead of ff800000 when the division of two operands leads to -inf .( say dividend is -2 and divisor is zero or a Denormal). I just wanted to know that does rounding affects infinity ,i.e, the upper/lower bounds . could you please let me know . thanks

aswaterman, 2020-02-23:
Rounding only affects whether the result becomes infinite if the unrounded result is greater than the largest representable value. It’s the usual IEEE 754 rules. Nothing here is RISC-V-specific.

smtdta, 2020-02-23:
[image]
Sir , the attached pic is when I ran soft float on RISC-V core ... in the first and the 4th case there is a mismatch , 

lets take case 1 : the operands are : -fe.7feffc (0xff7feffc) and -4c.003ff7 (0xa6003ff7) which makes the result to be in the order of ~10E54 which is way more than 3.4028235E38(largest representable value) . so , my core is giving +ff.000000 (+inf) where as Softfloat is expecting largest normal value . am I missing something ? why is there a  mismatch ? the same case repeats for rounding max , and min_mag cases too .

regards

foofoofoolishlearner, 2020-02-24:
The final result is related to the rounding mode.
If the result is +inf, the rounding direction is down，the final result should be 7f800000-1=7f7fffff. The same goes for -inf. If the rounding direction is up, the final result should be ff7fffff.

smtdta, 2020-02-24:
[[image] 2020-02-24 at 2 03 41 PM](https://user-images.githubusercontent.com/52655713/75137970-0815e500-570f-11ea-8b10-30f8602ca547.png)

right , I guess the source code (Softfloat) attached says so . but, am confused since I wasn't getting the same errors for Sqrt unit which is shared with division unit .

aswaterman, 2020-02-24:
I get the same result (0x7f7fffff) on my x86 laptop:

```
$ cat<<EOF | gcc -O2 test.c -lm && ./a.out
> #include <stdio.h>
> #include <fenv.h>
> int main()
> {
>   typedef union { int i; float f; } u;
>   volatile u a, b, c;
> 
>   a.i = 0xff7feffc;
>   b.i = 0xa6003ff7;
> 
>   fesetround(FE_DOWNWARD);
>   c.f = a.f / b.f;
>   printf("%x\n", c.i);
> 
>   return 0;
> }
> EOF
7f7fffff
```

smtdta, 2020-02-25:
@aswaterman  can you confirm whether other operations such as ADD , MUL , FUSED ADD/MUL fetches the same results . I mean, if one of the operand is inf (+/-) for add, multiplayer , sub, fused cases,  the same rationale of rounding infinity should hold as like it is in division . 

I'm, asking coz in my case , my RISC-V core doesn't round up/down inf  in any cases . it treats infinity as 7f800000 /ff800000 . so, do I need to change the RTL ? 

ill be grateful if could you please run the results for ADD, MUL, FUSED cases just like above on x86 laptop with one operand as +/- inf .

aswaterman, 2020-02-25:
Yes, that's right.  And since RISC-V uses IEEE 754 FP, yeah, you need to do that for RISC-V compliance.