riscv-software-src/riscv-isa-sim #373: Executing FDIV.S instruction in RV64G ISA

abdullahyildiz, 2019-12-31:
I couldn't find an explanation in the RISC-V ISA manual on how a RV64G ISA should execute a FDIV.S instruction on 64-bit floating-point registers.

Here is my example on executing this instruction:

`fdiv.s  fs4, ft8, ft1`

When I examine the Spike log, I noticed that registers ft8 and ft1 are both set to

_0x41d887d763c00000_

before executing 

`fdiv.s  fs4, ft8, ft1`

instruction.

In Spike simulation, this instruction calculates the division and finds the result as NaN (_0xffffffff7fc00000_).

Could you please explain why do I get such a result in Spike simulation?

I've attached the Spike log below.
[riscv_floating_point_arithmetic_test.0.log](https://github.com/riscv/riscv-isa-sim/files/4012608/riscv_floating_point_arithmetic_test.0.log)

luismarques, 2019-12-31:
Hello,

If you search the spec for NaN-boxing that will probably help clear the
confusion. In summary, the 32-bit float is stored in the space for a 64-bit
float, with the upper bits set to 1s (thus a NaN, when you look at the
whole 64 bits). The tools show you the whole 64-bit value instead of the
relevant lower 32 bits, which can be confusing but is correct. If you
extract the lower 32 bits and interpret that as a float then I expect the
value will make sense.

Cheers,
Luís

abdullahyildiz, 2019-12-31:
@luismarques,

Actually, my question is about how 

the division

```
ft8 = 0x41d887d763c00000;
ft1 = 0x41d887d763c00000;
fs4 = ft8/ft1;
```

results in NaN for the instruction

`fdiv.s fs4, ft8, ft1`.

luismarques, 2019-12-31:
From the spec:

> A narrow n-bit operation, where n <FLEN, checks that input operands are correctly NaN-boxed, i.e., all upper FLEN−n bits are 1.  If so, then least-significant bits of the input are used as the input value, otherwise the input value is treated as an n-bit canonical NaN.

abdullahyildiz, 2019-12-31:
@luismarques,

OK. Thanks for your help!

closed 2019-12-31 by abdullahyildiz