riscv-software-src/riscv-isa-sim #374: How to enable commit log for vector instructions

taoliug, 2020-01-02:
I built spike with --enable-commit, and I can see the commit log for non-vector instructions are generated properly, However, seems like there's no commit log for vector instructions. Is there any additional option needs to be enabled?

`core   0: 0x00000000800002b8 (0x77203f57) vmsle.vi v30, v18, 0
3 0x00000000800002b8 (0x77203f57)
core   0: 0x00000000800002bc (0x58eb734b) fnmsub.s ft6, fs6, fa4, fa1
3 0x00000000800002bc (0x58eb734b) f 6 0xffffffff7fc00000
core   0: 0x00000000800002c0 (0xc22d2657) vwaddu.vv v12, v2, v26
3 0x00000000800002c0 (0xc22d2657)
core   0: 0x00000000800002c4 (0x97802757) vmul.vv v14, v24, v0
3 0x00000000800002c4 (0x97802757)
core   0: 0x00000000800002c8 (0x10b8fe53) fmul.s  ft8, fa7, fa1
3 0x00000000800002c8 (0x10b8fe53) f28 0xffffffff7fc00000
core   0: 0x00000000800002cc (0xd2050053) fcvt.d.w ft0, a0
3 0x00000000800002cc (0xd2050053) f 0 0xc1dfffa0f2000000
core   0: 0x00000000800002d0 (0xa37c1dd3) flt.d   s11, fs8, fs7
3 0x00000000800002d0 (0xa37c1dd3) x27 0x0000000000000001
core   0: 0x00000000800002d4 (0x97240157) vsll.vv v2, v18, v8
3 0x00000000800002d4 (0x97240157)
`

chihminchao, 2020-01-02:
No, It is expected.  I have an commitlog prototype in my local branch but just for vector load/store unit. 
The output format is similar to current format.  The changes are
1. vector may has multiple register write
    multiple registers access for the same instruction are  listed in the same line with data length in vlen
2. vector may read or write memory by more than 1 times
    there are many read/write access record listed in the same line

Any suggestion is welcome.

taoliug, 2020-01-02:
Thanks for the reply. Sounds good to me as long as every architectural state is logged.